ros-industrial/ros_canopen #268: Add support for boolean datatype

ipa-mdl, 2018-02-14:
Support for boolean data could be added for SDOs.
However, the PDO granularity will be kept at 8, so single bits cannot be mapped for performance reasons.

The following should be suffiicient to get basic support:
1. add data type enum
2. add parser
3. add C type to lookup, `bool`?
4. prevent mapping of objects with type 1
5. testing

It should more or less straight-forward, but needs some testing with devices that offer bit-access.

rostest, 2019-02-18:
To support further can devices and eds-files for a customer, the boolean datatype is required soon. An implementation of 1-bit SDOs would be extremly helpfull. Would it be possible in the near future?

> It should more or less straight-forward, but needs some testing with devices that offer bit-access.

I have access to such a device and would like to help. If f.e. some beta-version is available for (pre-)testing, please give me a hint. Many thanks and best regards!

ipa-mdl, 2019-02-18:
>  Would it be possible in the near future?

Yes, if someone implements it. I don't have time (and no device) to do it.
That's why I added the "help wanted" label..

lkm1321, 2020-08-27:
Hi, I actually just added support for booleans in my fork, because my EDS file wasn't parsing because of boolean. I don't have a real device to test with at the moment, but I verified that:
1. EDS file containing booleans are parsed without throwing a std::bad_cast
2. When the user requests the object to be published, it's registered as an std_msgs/Bool topic

I'm happy to create a PR. Does it add on to melodic or melodic-devel?