ros-industrial/ros_canopen #271: CMake Error while making robotiq_s_model_control because of socketcan_interface

yamunamaccarana, 2018-03-19:
I have ros kinetic installed on Ubuntu 16.04.3 LTS.
I installed:
$ easy_install -U pymodbus
and:
$ git clone https://github.com/ros-industrial/robotiq.git
in my catkin_ws/src.
However, when I catkin_make, I get the error down below*.
I also tried to:
$ git clone https://github.com/ros-industrial/ros_canopen.git
and catkin_make, but this bring me to other errors with variables set to NOTFOUND and include folders.

What should I do?

* Here the error I get;
**-- +++ processing catkin package: 'robotiq_s_model_control'
-- ==> add_subdirectory(robotiq/robotiq_s_model_control)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /home/yami/catkin_ws/devel/share/socketcan_interface/cmake/socketcan_interfaceConfig.cmake:113 (message):
  Project 'socketcan_interface' specifies
  '/home/yami/catkin_ws/src/ros_canopen/socketcan_interface/include' as an
  include dir, which is not found.  It does neither exist as an absolute
  directory nor in
  '/home/yami/catkin_ws/src/ros_canopen/socketcan_interface//home/yami/catkin_ws/src/ros_canopen/socketcan_interface/include'.
  Check the issue tracker
  'https://github.com/ros-industrial/ros_canopen/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  robotiq/robotiq_s_model_control/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/yami/catkin_ws_man/build/CMakeFiles/CMakeOutput.log".
See also "/home/yami/catkin_ws_man/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed**

yamunamaccarana, 2018-03-19:
Probably it is better addressing this into https://github.com/ros-industrial/robotiq/issues/

ipa-mdl, 2018-03-19:
I cannot reproduce this.
I have added `robotiq` and `ros_canopen` to a clean workspace.
There are C++11 errors for `robotiq_s_model_articulated_gazebo_plugins`, but the rest compiles.

yamunamaccarana, 2018-03-20:
Thankyou,
If I try to compile both <code>robotiq</code> and <code>ros_canopen</code> together in a clean workspace I get:

-- +++ processing catkin package: 'robotiq_s_model_visualization'
-- ==> add_subdirectory(robotiq/robotiq_s_model_visualization)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
MUPARSER_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
   used as include directory in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
MUPARSER_LIBRARY (ADVANCED)
    linked by target "canopen_motor_node-test_muparser" in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node
    linked by target "canopen_motor" in directory /home/yami/catkin_ws_man/src/ros_canopen/canopen_motor_node

-- Configuring incomplete, errors occurred!
See also "/home/yami/catkin_ws_man/build/CMakeFiles/CMakeOutput.log".
See also "/home/yami/catkin_ws_man/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

And if I try to just cancel the <code>robotiq_s_model_visualization</code> the same error appears with other nodes in <code>robotiq</code>

ipa-mdl, 2018-03-20:
Did you install all requirements?
`rosdep install -i -y --from-paths /home/yami/catkin_ws_man/src`

It might be that after the CMakeLists.txt restructuring in `canopen_motor node` the missing dependency is not detected anymore.

yamunamaccarana, 2018-03-20:
Thank you ipa-mdl, this made my day.
Now I get other errors with <code>robotiq_s_model_articulated_gazebo_plugins</code>, as you told.
Thanks, again.

closed 2018-03-20 by yamunamaccarana

ipa-mdl, 2018-03-20:
I have now added a nicer error message (#272)