ros-industrial/ros_canopen #274: C++14 migration

ipa-mdl, 2018-03-31:
* [ ] move to `std` (requires #212) pointers and threads
* [ ] migrate to `std:function`, API compatible

JWhitleyAStuff, 2019-02-11:
@ipa-mdl - I proposed a different implementation of FastDelegate in #318. Thoughts as opposed to `std::function`?

ipa-mdl, 2019-02-12:
I chose FastDelegate because it was the fastest back then.. (and C++11 not yet allowed).
I'd like to migrate to `std::function` for the sake of maintenance and to get rid of this pointer magic ;)

JWhitleyAStuff, 2019-02-12:
I'll give the `std::function` version a try and see what I can do. I'll make a separate PR. Would you be willing to benchmark the results? For these use cases, I don't know that there will be much difference and `std::function` has better support.

JWhitleyAStuff, 2019-02-12:
@ipa-mdl See #318 and #320 for the alternate implementations. I couldn't find a good way to keep from breaking API for the *Delegate typedefs in the `std::function` version.