ros-industrial/ros_canopen #276: Schunk_lwa4d problem initialising with ros_canopen

ManolisCh, 2018-04-25:
Hi,

I am trying to use a schunk_lwa4d arm with an esd CAN-USB/2 in ROS but I have issues initializing it. I am using [this](https://github.com/ipa320/schunk_robots/blob/indigo_dev/schunk_lwa4d/launch/robot.launch) launch file in the shunck_lwa4p package from schunk_robots.

First I do: 

    sudo modprobe esd_usb2
    sudo ip link set can0 up type can bitrate 500000
    sudo ip link set can0 txqueuelen 20

and then I run the above launch file and try to initialise with `rosservice call /arm/driver/init`.
In the terminal I executed the rosservice I get:

    success: False
    message: "could not reset node '3'"

In the terminal I executed the ros launch I get:

    [ INFO] [1524494867.079043073]: Using fixed control period: 0.010000000
    [ INFO] [1524494871.222930279]: Initializing XXX
    [ INFO] [1524494871.223441822]: Current state: 1 device error: system:0 internal_error: 0 (OK)
    [ INFO] [1524494871.223800403]: Current state: 2 device error: system:0 internal_error: 0 (OK) error: 136
    [ INFO] [1524494871.224968675]: Current state: 2 device error: system:0 internal_error: 136 (protocol violations;)
    [ INFO] [1524494871.225023587]: Current state: 1 device error: system:0 internal_error: 136 (protocol violations;) ID: 136
    [ERROR] [1524494871.234310747]: CAN not ready
    [ INFO] [1524494871.899096913]: waitForService: Service [/arm/controller_manager/load_controller] has not been advertised, waiting...
    [ INFO] [1524494876.910831528]: waitForService: Service [/arm/controller_manager/load_controller] has not been advertised, waiting...
    [ INFO] [1524494881.225471058]: Current state: 1 device error: system:125 internal_error: 136 (protocol violations;)
    [ INFO] [1524494881.225579044]: Current state: 0 device error: system:125 internal_error: 136 (protocol violations;)
    [ INFO] [1524494881.225706891]: Current state: 0 device error: system:0 internal_error: 136 (protocol violations;)
    [ INFO] [1524494881.225850755]: Current state: 0 device error: system:0 internal_error: 136 (protocol violations;)
    [ INFO] [1524494881.917377871]: waitForService: Service [/arm/controller_manager/load_controller] has not been 
    advertised, waiting...

Any ideas?

gavanderhoorn, 2018-04-25:
In case of [cross-posts](https://answers.ros.org/question/289428) please always mention that you did that so we can avoid split discussions and duplication of effort. Thanks.

ManolisCh, 2018-04-25:
yeap your right, I just edited both of my posts to mention the crossposting!

ipa-mdl, 2018-04-25:
I am pretty sure that this is a configuration issue.
The config and launch files you are using are not part of this repository, so I will close this issue here in favor of the ROS answers post.

closed 2018-04-25 by ipa-mdl