ros-industrial/ros_canopen #288: Melodic release

ipa-mdl, 2018-06-08:
`melodic_devel` whould work, but needs some testing because of the interface chanages.

kmhallen, 2018-06-26:
In the short term, could you bloom/release only the ```can_msgs``` package?

There are several packages that only depend on the ```can_msgs``` package that are waiting for its release:
http://wiki.ros.org/can_msgs?distro=kinetic

This could be implemented by creating a ```melodic.ignored``` file in the release repository temporarily:
https://github.com/DataspeedInc-release/dataspeed_can-release/blob/master/melodic.ignored
```
canopen_402
canopen_chain_node
canopen_master
canopen_motot_node
ros_canopen
socketcan_bridge
socketcan_interface
```

ipa-mdl, 2018-07-10:
> This could be implemented by creating a melodic.ignored file in the release repository temporarily:

Yes, but that would force me to tag a version for all packages in the source repository.

> There are several packages that only depend on the can_msgs package that are waiting for its release

I plan to test and release on [WRID18](https://rosindustrial.org/events/2018/7/10/world-ros-i-day).

ipa-mdl, 2018-07-11:
tested and released: https://github.com/ros/rosdistro/pull/18444

closed 2018-07-11 by ipa-mdl