ros-industrial/ros_canopen #293: Sending RPDO and SYNC message with hard-real time constraints- is it possible?

mohse-n, 2018-07-09:
I would like to use ros_canopen for giving position to a Delta servo drive in interpolation mode, with rates of up to 1 KHz. But there is a caveat: For every position sent via RPDO, a SYNC frame must also be sent from master. If the drive doesn't receive a SYNC message within 2*update interval (2 ms in case of an update rate of 1 KHz) it will give an alarm (and stop working). In other words, there is a hard real-time deadline for sending SYNC messages.
1- Can I **reliably** do this with `ros_canopen` using a PREEMPT_RT patched linux? 
2- Slightly unrelated to this: are there any example for getting started with `ros_canopen`?

ipa-mdl, 2018-07-10:
I have never tried to use it with PREEMPT_RT patch or higher rates than 250 Hz.

> 1- Can I reliably do this with ros_canopen using a PREEMPT_RT patched linux?

For just sending the SYNC, I would say yes.
`canopen_chain_node` contains a `sync_node`, which uses the `BCM` feature of the kernel that might work at 1000 Hz.

To use it together with the chain nodes (=sending commands), the `canopen::ExternalMaster` plugin needs to be enabled (or a custom plugin).
Please note, this plugin and the rest of the chain are not realtime-safe due to mutlthreading, mutexes, etc..
The primary intention for `ros_canopen` was flexibility.

> 2- Slightly unrelated to this: are there any examples for getting started with ros_canopen?

Currently there is now proper guide, I would recommend to have a look at ROS answers.

closed 2018-08-23 by mohse-n

aatb-ch, 2022-04-17:
Hello, I have a related question to know the current status of realtime capabilities of the `canopen_motor_node`.

I have a Festo CMMP drive in interpolation mode at 100Hz, it works perfectly on a Raspi 4, but I have had on very rare cases jittery motion when the drive looses sync and the servo will vibrate very badly until catching up again. I suspect that the OS is doing some housekeeping that delays some SYNC frames and causes synchronization error. I have set the drive to trigger a QStop in case of lost sync to avoid any potential damage to the process, but the CMMP drives do not allow a larger interpolation time that 10ms and it seems to be slightly thin margin with non-RT OS latency.

I have been running PREEMPT_RT kernels for our UR robots and this has been night and day in the stability of the communication so I believe it would help mitigate this problem. Have the current realtime capabilities in `ros_canopen` changed since the above mentionned advice?