ros-industrial/ros_canopen #297: Cannot run multiple concurrent instances of the socketcan_to_topic node

tge-rct, 2018-07-26:
When using CAN interfaces with multiple CAN channels, the running of a second socketcan_to_topic node instance breaks the first, probably due to a node name collision.
to reproduce:
`rosrun socketcan_bridge socketcan_to_topic_node _can_device:=can0`
then
`rosrun socketcan_bridge socketcan_to_topic_node _can_device:=can1`

output from can0 node:

```
[ INFO] [1532596865.868098784]: Successfully connected to can0.
[ WARN] [1532596909.997854110]: Shutdown request received.
[ WARN] [1532596909.997935854]: Reason given for shutdown: [new node registered with same name
```
I will look at a fix shortly

tge-rct, 2018-07-26:
Does not work:

```
tom@poly:~$ rosrun socketcan_bridge socketcan_to_topic_node __name=node0 _can_device:=can0 &  
[1] 3293                                                                                                           
tom@poly:~$ [ INFO] [1532600602.055855235]: Successfully connected to can0.                    
rosrun socketcan_bridge socketcan_to_topic_node __name=node1 _can_device:=can1 &                                   
[2] 3324                                                                                                           
tom@poly:~$ [ INFO] [1532600608.935094376]: Successfully connected to can1.                    
[ WARN] [1532600608.995047916]: Shutdown request received.                                                         
[ WARN] [1532600608.995125580]: Reason given for shutdown: [new node registered with same name]                    
                                                                                                                   
[1]-  Done                    rosrun socketcan_bridge socketcan_to_topic_node __name=node0 _can_device:=can0       
tom@poly:~$                                                                                    
                                
```                                                                                  

Probably due to :
https://github.com/ros-industrial/ros_canopen/blob/410984dc5ec8be0e41686a42618fdfa7c037e570/socketcan_bridge/src/socketcan_to_topic_node.cpp#L38

gavanderhoorn, 2018-07-26:
You're missing a colon between `__name` and `=node0`.

ipa-mdl, 2018-07-26:
@gavanderhoorn is right.

 > Edit: I would recommend to use a launch file for this btw.

This was my first thought as well.

I will close this issue now, because it is no issue with `ros_canopen`, but just how ROS works.

closed 2018-07-26 by ipa-mdl