ros-industrial/ros_canopen #299: class_loader dependencies error

jacknlliu, 2018-09-17:
Compile this package using the kinetic branch, but I got the `class_loader/class_loader.hpp` not found.

```sh
/catkin_ws/src/ros_canopen/socketcan_interface/src/socketcan_interface_plugin.cpp:1:42: fatal error: class_loader/class_loader.hpp: No such file or directory
compilation terminated.
ros_canopen/socketcan_interface/CMakeFiles/socketcan_interface_plugin.dir/build.make:62: recipe for target 'ros_canopen/socketcan_interface/CMakeFiles/socketcan_interface_plugin.dir/src/socketcan_interface_plugin.cpp.o' failed
make[2]: *** [ros_canopen/socketcan_interface/CMakeFiles/socketcan_interface_plugin.dir/src/socketcan_interface_plugin.cpp.o] Error 1
CMakeFiles/Makefile2:444: recipe for target 'ros_canopen/socketcan_interface/CMakeFiles/socketcan_interface_plugin.dir/all' failed
```

Then
```sh
rosdep install -i -y --from-paths src  --rosdistro kinetic -r

#All required rosdeps installed successfully
```

And I have installed the system package `libclass-loader-dev` version `0.3.2-1`.

Any help will be appreciated!

jacknlliu, 2018-09-17:
After upgrade the ros package `ros-kinetic-class-loader` to the latest version, it works! But why this problem occurs before upgrading the pacakge to the latest version.

ipa-mdl, 2018-09-17:
Six months ago the `class_loader` package migrated to *.hpp file names (https://github.com/ros/class_loader/pull/84).

closed 2018-09-17 by ipa-mdl