ros-industrial/ros_canopen #300: controller_manager services not advertised after node `canopen_motor_node`?

jacknlliu, 2018-09-23:
I run the following launch file,
```xml
<node ns="arm" name="driver" pkg="canopen_motor_node" type="canopen_motor_node" output="screen" clear_params="true" launch-prefix="gdb -ex run --args">
  <!--<node ns="arm" name="driver" pkg="canopen_motor_node" type="canopen_motor_node" output="screen" clear_params="true" launch-prefix="">-->
     <rosparam command="load" file="$(find rgm_driver)/config/can0.yaml" />
     <rosparam command="load" file="$(find rgm_driver)/config/arm_driver.yaml" />
</node>

<!-- Load controller settings -->
<rosparam ns="arm" command="load" file="$(find rgm_driver)/config/joint_state_controller.yaml"/>
<rosparam ns="arm" command="load" file="$(find rgm_driver)/config/arm_controller.yaml"/>

<!-- spawn controller manager -->
<node ns="arm" name="ros_control_controller_spawner" pkg="controller_manager" type="controller_manager" respawn="false"
    output="screen" args="spawn joint_state_controller" />

<!-- load controllers using cob -->
<node ns="arm" name="cob_control_mode_adapter_node" pkg="cob_control_mode_adapter" type="cob_control_mode_adapter_node" cwd="node" respawn="false" output="screen"/>
```

And the output is:

```
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
process[arm/ros_control_controller_spawner-3]: started with pid [3672]
process[arm/cob_control_mode_adapter_node-4]: started with pid [3678]
[ INFO] [1537716055.826437993]: waitForService: Service [/arm/controller_manager/load_controller] has not been advertised, waiting...
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fffedf1d700 (LWP 3708)]
[New Thread 0x7fffed71c700 (LWP 3709)]
[New Thread 0x7fffecf1b700 (LWP 3710)]
[New Thread 0x7fffe7fff700 (LWP 3715)]
[New Thread 0x7fffe77fe700 (LWP 3716)]
[New Thread 0x7fffe6ffd700 (LWP 3717)]
[New Thread 0x7fffe67fc700 (LWP 3718)]
[New Thread 0x7fffe5ffb700 (LWP 3719)]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[New Thread 0x7fffe57fa700 (LWP 3727)]
[New Thread 0x7fffe4ff9700 (LWP 3732)]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
[tcsetpgrp failed in terminal_inferior: Inappropriate ioctl for device]
Did not receive a response message
[ INFO] [1537716057.371318600]: Using fixed control period: 0.010000000
[ INFO] [1537716060.826751596]: waitForService: Service [/arm/controller_manager/load_controller] has not been advertised, waiting...
[ INFO] [1537716065.836552990]: waitForService: Service [/arm/controller_manager/load_controller] has not been advertised, waiting...
```

The controller can't load because the `controller_manager/load_controller` has not been advertised. It means that the `controller_manager` is not Instantiated in the node `canopen_motor_node`?

ipa-mdl, 2018-09-23:
These services will appear after you have called the `driver/init` service.

ipa-mdl, 2018-09-23:
>  This is very different from other ROS driver which will enable load and start controllers using the controller manager after starting the driver.

This has been the default for all Care-O-bot drivers..
Most drivers do not support homing, which is the one of reasons for this behavior.
The controllers will start as soon as you have called init.
Of course, you can that call to your launchffile, either with `rosservice` or a custom script.

> It will be better if a document about this can be provided in ros wiki.

The services are described in [here](http://wiki.ros.org/canopen_chain_node#Services)
I would appriciate if you could add whatever you think is missing :)

closed 2019-01-16 by ipa-mdl