ros-industrial/ros_canopen #301: Error output with no explicit meaning when init the driver

jacknlliu, 2018-09-24:
After start the ros node `canopen_motor_node`, I run the initial driver
```sh
rosservice call /arm/driver/init
```
The output is
```sh
success: False
message: "Transition timeout; Could not enable motor"
```
And the output from `canopen_motor_node` is:
```
[ INFO] [1537756949.778140475]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[New Thread 0x7fffc7547700 (LWP 9656)]
[ INFO] [1537756949.778630198]: Current state: 2 device error: system:0 internal_error: 0 (OK)
[New Thread 0x7fffc6d46700 (LWP 9657)]
[ INFO] [1537756952.890460416]: waitForService: Service [/arm/controller_manager/load_controller] has not been advertised, waiting...
[ INFO] [1537756956.795575493]: Current state: 2 device error: system:125 internal_error: 0 (OK)
[ INFO] [1537756956.795698509]: Current state: 0 device error: system:125 internal_error: 0 (OK)
[ INFO] [1537756956.795766744]: Current state: 0 device error: system:0 internal_error: 0 (OK)
[Thread 0x7fffc6d46700 (LWP 9657) exited]
[ INFO] [1537756956.795854967]: Current state: 0 device error: system:0 internal_error: 0 (OK)
[Thread 0x7fffc7547700 (LWP 9656) exited]
[Thread 0x7fffc7d48700 (LWP 9655) exited]
```

Any help will be appreciated!

ipa-mdl, 2018-09-24:
> success: False
message: "Transition timeout; Could not enable motor"

That's quite explicit. 

> Any help will be appreciated!

This is just an issue tracker.
Please use [ROS answers](https://answers.ros.org) if you need support.
There are plenty of entries regarding `ros_canopen` and different hardware set-ups.
If you want to increase your chance of getting a helpful answer over there, you should include more information with your question:
* Which version are you using?
* What device do you trying to connect with?
* Link to launch files and config

closed 2018-09-24 by ipa-mdl

jacknlliu, 2018-09-24:
moved to https://answers.ros.org/question/304108/ros_canopen-driver-init-failed-with-kollmorgen-rgm/

jacknlliu, 2018-09-27:
@ipa-mdl  We find that the driver did't send any control word when the driver needs switch the mode(`driver/init` service call), but give the error `Transition timeout`. 

We monitor the can0 by `candump`:
```sh
  can0  000   [2]  82 01
  can0  701   [1]  00
  can0  601   [8]  2B 17 10 00 00 00 00 00
  can0  581   [8]  60 17 10 00 00 00 00 00
  can0  601   [8]  40 41 60 00 00 00 00 00
  can0  581   [8]  4B 41 60 00 70 06 00 00
  can0  601   [8]  40 61 60 00 00 00 00 00
  can0  581   [8]  4F 61 60 00 06 06 00 00
  can0  601   [8]  40 64 60 00 00 00 00 00
  can0  581   [8]  43 64 60 00 B7 41 75 00
  can0  601   [8]  40 6C 60 00 00 00 00 00
  can0  581   [8]  43 6C 60 00 00 00 00 00
  can0  601   [8]  40 77 60 00 00 00 00 00
  can0  581   [8]  4B 77 60 00 00 00 00 00
  can0  601   [8]  40 FD 60 00 00 00 00 00
  can0  581   [8]  43 FD 60 00 08 00 00 00
  can0  601   [8]  40 40 60 00 00 00 00 00
  can0  581   [8]  4B 40 60 00 04 00 00 00
  can0  601   [8]  40 60 60 00 00 00 00 00
  can0  581   [8]  4F 60 60 00 06 00 00 00
  can0  601   [8]  40 7A 60 00 00 00 00 00
  can0  581   [8]  43 7A 60 00 00 00 08 00
  can0  601   [8]  40 FF 60 00 00 00 00 00
  can0  581   [8]  43 FF 60 00 00 00 00 00
  can0  601   [8]  40 71 60 00 00 00 00 00
  can0  581   [8]  4B 71 60 00 00 00 00 00
  can0  601   [8]  2F 04 18 02 01 00 00 00
  can0  581   [8]  60 04 18 02 00 00 00 00
  can0  601   [8]  2F 03 18 02 01 00 00 00
  can0  581   [8]  60 03 18 02 00 00 00 00
  can0  601   [8]  2F 01 18 02 01 00 00 00
  can0  581   [8]  60 01 18 02 00 00 00 00
  can0  601   [8]  2F 05 18 02 01 00 00 00
  can0  581   [8]  60 05 18 02 00 00 00 00
  can0  601   [8]  2F 06 18 02 01 00 00 00
  can0  581   [8]  60 06 18 02 00 00 00 00
  can0  601   [8]  2F 02 18 02 01 00 00 00
  can0  581   [8]  60 02 18 02 00 00 00 00
  can0  601   [8]  2F 05 14 02 01 00 00 00
  can0  581   [8]  60 05 14 02 00 00 00 00
  can0  601   [8]  2F 03 14 02 01 00 00 00
  can0  581   [8]  60 03 14 02 00 00 00 00
  can0  601   [8]  2F 02 14 02 01 00 00 00
  can0  581   [8]  60 02 14 02 00 00 00 00
  can0  601   [8]  2F 01 14 02 01 00 00 00
  can0  581   [8]  60 01 14 02 00 00 00 00
  can0  601   [8]  2F 07 18 02 01 00 00 00
  can0  581   [8]  60 07 18 02 00 00 00 00
  can0  601   [8]  2F 00 18 02 01 00 00 00
  can0  581   [8]  60 00 18 02 00 00 00 00
  can0  601   [8]  2F 06 14 02 01 00 00 00
  can0  581   [8]  60 06 14 02 00 00 00 00
  can0  601   [8]  2F 07 14 02 01 00 00 00
  can0  581   [8]  60 07 14 02 00 00 00 00
  can0  601   [8]  2F 04 14 02 01 00 00 00
  can0  581   [8]  60 04 14 02 00 00 00 00
  can0  601   [8]  2F 00 14 02 01 00 00 00
  can0  581   [8]  60 00 14 02 00 00 00 00
  can0  000   [2]  01 01
  can0  080   [0] 
  can0  181   [2]  70 06
  can0  281   [3]  70 06 06
  can0  381   [6]  70 06 B1 41 75 00
  can0  481   [6]  70 06 00 00 00 00
```