ros-industrial/ros_canopen #304: Socketbridge_socket_to_can Regarding

lakshmanb09031992, 2018-10-02:
Hello,

I am using Peak USB2CAN interface to get sensor information into ROS environment.

I am able to use socket_interface and check the frames via socket_dump. But when i use socket_bridge to convert the frames to topics i am getting a segmentation fault (core dumped) error.

[image]

Is this something related to the hardware i use or something else. I am using ROS kinetic distribtion in Ubuntu 16.04 system. Thank you in advance

ipa-mdl, 2018-10-03:
Please make sure that you have upgraded all packages to the latest version.
For further debugging please prefix it with gdb to get a stack trace.

lakshmanb09031992, 2018-10-04:
Thank you upgrading solved the error