ros-industrial/ros_canopen #305: No PDO frames were sent and Transition timeout

jacknlliu, 2018-10-03:
Using `rosservice call /arm/driver/init` after start the `canopen_motor_node`, and we get an error
```
success: False
message: "Transition timeout; Could not enable motor"
```

The output of `canopen_motor_node` like

```
[Thread debugging using libthread_db enabled]
[New Thread 0x7fffedf1d700 (LWP 26114)]
[New Thread 0x7fffed71c700 (LWP 26115)]
[New Thread 0x7fffecf1b700 (LWP 26116)]
[New Thread 0x7fffe7fff700 (LWP 26121)]
[New Thread 0x7fffe77fe700 (LWP 26123)]
[New Thread 0x7fffe6ffd700 (LWP 26124)]
[New Thread 0x7fffe67fc700 (LWP 26125)]
[New Thread 0x7fffe5ffb700 (LWP 26126)]
[New Thread 0x7fffe57fa700 (LWP 26132)]
[New Thread 0x7fffe4ff9700 (LWP 26137)]
Did not receive a response message
[ INFO] [1538577922.610506074]: Using fixed control period: 0.010000000
[/arm/controller_manager/load_controller] has not been advertised, waiting...
[ INFO] [1538577924.365880490]: Initializing XXX
[New Thread 0x7fffcbd42700 (LWP 26318)]
[ INFO] [1538577924.366409816]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[New Thread 0x7fffcb541700 (LWP 26319)]
[ INFO] [1538577924.366723095]: Current state: 2 device error: system:0 internal_error: 0 (OK)
[New Thread 0x7fffcad40700 (LWP 26320)]
[ INFO] [1538577929.282573121]: waitForService: Service [/arm/controller_manager/load_controller] has not been advertised, waiting...
[ INFO] [1538577931.378695364]: Current state: 2 device error: system:125 internal_error: 0 (OK)
[ INFO] [1538577931.378851047]: Current state: 0 device error: system:125 internal_error: 0 (OK)
[ INFO] [1538577931.379019644]: Current state: 0 device error: system:0 internal_error: 0 (OK)
[Thread 0x7fffcad40700 (LWP 26320) exited]
[ INFO] [1538577931.379457254]: Current state: 0 device error: system:0 internal_error: 0 (OK)
[Thread 0x7fffcb541700 (LWP 26319) exited]
[Thread 0x7fffcbd42700 (LWP 26318) exited]
```

We find that there are no PDO frames were sent to the CAN bus. I think no matter the transition success or failed, there should be a PDO frame sent to the bus.

And we can easily send the CAN frame to bus and enable the motor using `can-utils` to operate the RPDOs like
```sh
cansend can0 000#0100

cansend can0 201#0600
cansend can0 080#

cansend can0 201#0700
cansend can0 080#

cansend can0 201#0F00
cansend can0 080#
```

The following are the `candump` about `canopen_motor_node`
```sh
 can0  601   [8]  40 41 60 00 00 00 00 00
  can0  581   [8]  4B 41 60 00 70 06 00 00
  can0  601   [8]  40 61 60 00 00 00 00 00
  can0  581   [8]  4F 61 60 00 06 06 00 00
  can0  601   [8]  40 64 60 00 00 00 00 00
  can0  581   [8]  43 64 60 00 56 0F E5 FF
  can0  601   [8]  40 6C 60 00 00 00 00 00
  can0  581   [8]  43 6C 60 00 00 00 00 00
  can0  601   [8]  40 77 60 00 00 00 00 00
  can0  581   [8]  4B 77 60 00 FD FF 00 00
  can0  601   [8]  40 FD 60 00 00 00 00 00
  can0  581   [8]  43 FD 60 00 08 00 00 00
  can0  601   [8]  40 40 60 00 00 00 00 00
  can0  581   [8]  4B 40 60 00 04 00 00 00
  can0  601   [8]  40 60 60 00 00 00 00 00
  can0  581   [8]  4F 60 60 00 06 00 00 00
  can0  601   [8]  40 7A 60 00 00 00 00 00
  can0  581   [8]  43 7A 60 00 00 00 08 00
  can0  601   [8]  40 FF 60 00 00 00 00 00
  can0  581   [8]  43 FF 60 00 00 00 00 00
  can0  601   [8]  40 71 60 00 00 00 00 00
  can0  581   [8]  4B 71 60 00 00 00 00 00
  can0  000   [2]  01 01
  can0  080   [0] 
  can0  181   [3]  70 06 06
  can0  281   [5]  55 0F E5 FF 06
  can0  381   [8]  00 00 00 00 55 0F E5 FF
  can0  481   [6]  00 00 00 00 00 00
  can0  080   [0] 
  can0  181   [3]  70 06 06
  can0  281   [5]  57 0F E5 FF 06
  can0  381   [8]  00 00 00 00 57 0F E5 FF
  can0  481   [6]  03 00 00 00 00 00
  can0  080   [0] 
  can0  181   [3]  70 06 06
  can0  281   [5]  52 0F E5 FF 06
  can0  381   [8]  00 00 00 00 52 0F E5 FF
  can0  481   [6]  FF FF 00 00 00 00
```

ipa-mdl, 2018-10-03:
Please check your PDO settings.

jacknlliu, 2018-10-03:
Our PDO settings:
```
RPDO1(CoB-ID 201): 0x6040(control word), 0x6060(mode of operation).
RPDO2(CoB-ID 301): 0x607A(target position), 0x6081(profile velocity)
RPDO3(CoB-ID 401): 0x60FF(Target Velocity)
RPDO4(CoB-ID 501): 0x6071(target torque)

TPDO1(CoB-ID 181): 0x6041 (status word), 0x6061 (mode of operation)
TPDO2(CoB-ID 281): 0x6064 (actual position)
TPDO3(CoB-ID 381): 0x606C (actual velocity)
TPDO4(CoB-ID 481): 0x6077 (actual torque)
```
We can use this configuration and send PDO frames using `can-utils` to motors.

jacknlliu, 2018-10-03:
Any tips can help for debugging this?

ipa-mdl, 2018-10-03:
> We can use this configuration and send PDO frames using can-utils to motors.

This does not matter. You must configure it in the EDS.

jacknlliu, 2018-10-03:
Yeah, I configure it in the EDS, and the EDS is [here](https://gist.github.com/jacknlliu/fd9c6632c53cc0bfbbacc5e0c62317d3).

jacknlliu, 2018-10-03:
I change the EDS file, but get the following error
```
abort1001#0, reason: Object does not exist in the object dictionary.
```
How to find which object not exist in the object dictionary?

jacknlliu, 2018-10-04:
Our drive doesn't support this object(0x1001). I just comment the following lines 

https://github.com/ros-industrial/ros_canopen/blob/410984dc5ec8be0e41686a42618fdfa7c037e570/canopen_master/src/sdo.cpp#L358

Is there an easy way to prevent querying this object(0x1001, error register)?

closed 2019-01-16 by ipa-mdl