ros-industrial/ros_canopen #306: Using technical terms incorrectly?

jacknlliu, 2018-10-03:
https://github.com/ros-industrial/ros_canopen/blob/410984dc5ec8be0e41686a42618fdfa7c037e570/canopen_master/src/pdo.cpp#L179

https://github.com/ros-industrial/ros_canopen/blob/410984dc5ec8be0e41686a42618fdfa7c037e570/canopen_master/src/pdo.cpp#L190

Why use the CAN `TPDO` parameters to create a object named `RPDO`, and why use the CAN `RPDO` parameters to create a object named `TPDO`?  @ipa-mdl

ipa-mdl, 2018-10-03:
A RPDO on one side is a TPDO on the other and vice versa.

jacknlliu, 2018-10-04:
@ipa-mdl  Should not use this transformation! If they appear at the same time, they will be difficult to distinguish, and it will be not easy to debug these codes!

ipa-mdl, 2018-10-04:
I don't understand your comment.

The local receiver (-> class instance) binds to the TPDOs specified in the remote dictionary (-> Macro).
It's confusing at first, but this is how it works.
Have you ever wired a serial bus? ;)

jacknlliu, 2018-10-04:
I think TPDO and RPDO are specified in CANopen, and they have the unique meaning in CAN. TPDO can be named as receiver/reader, and RPDO can be named as sender/writer, but name `TPDO` as RPDO, `RPDO` as `TPDO` is not a good idea.

ipa-mdl, 2018-10-05:
The `RPDO` class consumes PDO data, the `TPDO` class produces PDO data (which reflects the definition in CiA 301).
If these classes would be used in device context, `RPDO` class would get  configured by `RPDO` entries in the object dictionary.

However, the `PDOMapper` (as-is) deals with remote objects, so the `TPDO` gets configured by `RPDO` entrires and vice versa.
I do not see the need to change this anytime soon.

Of course, this needs better documentation and help would be very much appreciated.

closed 2019-01-16 by ipa-mdl