ros-industrial/ros_canopen #311: hardware_interface::PositionJointInterface cannot be provided in mode 7

CccYunxiao, 2018-11-26:
I want to use Interpolated Position mode to motors ｍｏｔｉｏｎ, ｓｏ i use the node canopen_motor_node with operation modes ID.7

There is an error：　hardware_interface::PositionJointInterface cannot be provided in mode 7

Ｓｏ hardware_interface::PositionJointInterface can not accept IP mode(ID.7) ? Right?

Who can enlighten me? Thx very much!!

CccYunxiao, 2018-11-26:
@tfoote @ipa-mdl

CccYunxiao, 2018-11-26:
@floweisshardt

CccYunxiao, 2018-11-26:
joint_trajectory_controller:
  type: position_controllers/JointTrajectoryController
  topic: "test"
  joints:
    - rig1_plate_joint
  constraints:
    goal_time: 0.6
    stopped_velocity_tolerance: 0.5
    rig1_plate_joint: {trajectory: 0.1, goal: 0.1}
  stop_trajectory_duration: 0.5
  state_publish_rate:  25
  action_monitor_rate: 10
  required_drive_mode: 7

ipa-mdl, 2018-11-26:
This is an issue tracker and no support forum.
In addtion `canopen_test_utils` is mostly for internal usage and intentionally not part of this repository.
If you need help in setting up your device, you may find a solution on [ROS answers](https://answers.ros.org).
If none of the existing answers work for you, you can ask specificially over there.

Please do not tag others here, especially if they are unrelated.
Thanks!

closed 2018-11-26 by ipa-mdl