ros-industrial/ros_canopen #314: canopen_master can not find library '-lpthread'

tommy3001, 2019-01-02:
The error message is: 
`CMake Error at /home/tommy/dev/catkin_ws/devel/share/socketcan_interface/cmake/socketcan_interfaceConfig.cmake:148 (message):
  Project 'canopen_master' tried to find library '-lpthread'.  The library is
  neither a target nor built/installed properly.  Did you compile project
  'socketcan_interface'? Did you find_package() it before the subdirectory
  containing its code is included?
`
I have an arch linux installation and in line 119 in devel/share/socketcan_interface/cmake/socketcan_interfaceConfig.cmake 
I have 
set(libraries "socketcan_interface_string;/usr/lib/libboost_chrono.so;/usr/lib/libboost_system.so;/usr/lib/libboost_thread.so;-lpthread;/usr/lib/libboost_date_time.so;/usr/lib/libboost_atomic.so")

When I change -lpthread to /usr/lib/libpthread.so, it will be overwritten in the next run of catkin_make

gavanderhoorn, 2019-01-02:
Please see [#q311685](https://answers.ros.org/question/311685) on ROS Answers, and ultimately https://github.com/ros/ros-overlay/issues/711.

ipa-mdl, 2019-01-02:
As @gavanderhoorn already pointed out, this seems to be problem with a [change in CMake](https://gitlab.kitware.com/cmake/cmake/commit/bd831ed0948a1e99f573f0056f2bee5d3b21009e).
This must be fixed in CMake or catkin (https://github.com/ros/catkin/pull/975).

JWhitleyAStuff, 2019-03-15:
@ipa-mdl Is there any reason we shouldn't migrate to std::threads instead of boost::thread? Wouldn't that solve this problem? It should be feasible going forward since C++11 threads were well-supported in gcc that is included with Ubuntu 16.04 and Indigo will be EOL next month.

ipa-mdl, 2020-07-31:
Was fixed in catkin.

closed 2020-07-31 by ipa-mdl