ros-industrial/ros_canopen #315: Motors do not start at same time

jcrobots, 2019-01-12:
I am using ROS_CANOpen to move a differential drive robot, with a config setup based on: https://github.com/ipa320/schunk_robots/tree/indigo_dev/schunk_lwa4p 

I'm using profiled position mode and publishing to the topic "joint_group_position_controller/command" to send target positions.  I have found that sometimes when I publish a command, one of the two wheel motors will begin moving one SYNC interval later than the other (I am having to use a 200ms SYNC interval due to this issue: https://github.com/ipa320/schunk_robots/issues/59).  I have been told that if I use SYNC messages and my two servo drives have the same group identifier (in firmware they both have group ID 128), they should start at the same time.  However there does not seem to be any way to specify a group ID in ROS_CANOpen.  It may be notable that I have been unable to reproduce this issue using profiled velocity mode.  

I believe this may be a missing functionality in ROS_CANOpen, but if this can be fixed on my end, I would very much appreciate any help or advice!

Thank you for reading

jcrobots, 2019-01-12:
Correct, is that ok?

ipa-mdl, 2019-01-16:
I addressed most of the question on ROS answers.

> my two servo drives have the same group identifier (in firmware they both have group ID 128),

I don't know what a "group identifier" should be. It looks like the SYNC object (128 = 0x80).
At the moment this value is [hard-coded](https://github.com/ros-industrial/ros_canopen/blob/162c0a8b20fda2f2a4b13ab1c26e1437beeacdb3/canopen_chain_node/src/ros_chain.cpp#L328) to this value, it should be synchronous then.

> is that ok?

Not really, I will close this issue in favor of ROS answers.

closed 2019-01-16 by ipa-mdl