ros-industrial/ros_canopen #319: Log error message, if initialization / recover fails

JonathanGruner, 2019-02-12:
When calling the service interface to init / recover the motor, there is an error string in the response of the service call, if something fails. Calling the service from the launch file on the other hand does not give any error indication to the user.

There should be the same error string printed to the console:

```
[ INFO] [1549975063.533360109]: Initializing XXX
[ INFO] [1549975063.533741997]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[ INFO] [1549975063.534281762]: Current state: 2 device error: system:0 internal_error: 0 (OK)
--> Error, but no warning / error message
```

ipa-mdl, 2019-02-12:
Thanks for reporting!
It should be quite easy to add this in `RosChain::handle_init` & Co.
I could imaging some scoped response logger.

jschleicher, 2019-07-04:
@ipa-mdl
Following the discussion in https://github.com/ros-industrial/ros_canopen/pull/286#discussion_r299515309, a plain `ROS_WARN_*` output in https://github.com/ros-industrial/ros_canopen/blob/6890527ee9af45afd8c607ff86b92e699de79fd4/canopen_chain_node/src/ros_chain.cpp#L108-L109 would be alright, correct? What is the diagnostics logger in canopen_chain_node about?

ipa-mdl, 2019-07-04:
> a plain ROS_WARN

`ROS_WARN` or `ROS_ERROR`.

> What is the diagnostics logger in canopen_chain_node about?

In general?

PR #356 (2019-10-01): Log services
closes #319, replaces #343 

@jschleicher: please give this a try
closed 2019-10-01 by ipa-mdl