ros-industrial/ros_canopen #323: exception when parse object dictionary

jacknlliu, 2019-03-12:
```sh
src/ros_canopen/canopen_master/include/canopen_master/objdict.h(227): Throw in function const EntryConstSharedPtr& canopen::ObjectDict::at(const canopen::ObjectDict::Key&) const
Dynamic exception type: boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<std::out_of_range> >
std::exception::what: Unable to find key in unordered_map.
[canopen::tag_objectdict_key*] = 6061
```

EDS file about `0x6061`:
```
[6061]
ParameterName=Modes of Operation Display
ObjectType=7
AccessType=RO
DataType=0x0003
PDOMapping=1
ObjFlags=0x00000000
LowLimit=-128
HighLimit=127
DefaultValue=1

[1A00sub2]
ParameterName=PDO Mapping Entry 2
ObjectType=7
AccessType=RW
DataType=0x0007
PDOMapping=0
ObjFlags=0x00000000
DefaultValue=0x60610010
```

jacknlliu, 2019-03-13:
Solved.

closed 2019-03-13 by jacknlliu

JadTawil-theonly, 2020-05-22:
what was the solution??

augenstern-M, 2022-09-12:
How to solve this problem?