ros-industrial/ros_canopen #324: Segfault during shutdown

jschleicher, 2019-03-19:
Sometimes I get a segfault during shutdown of `canopen_motor_node`. I could not really trace it down; please find a screenshot of the ubuntu crash-report attached.

[image]

I'll build with DebugSymbols from source and report here, if I can get a backtrace.

ipa-hsd, 2019-03-20:
Can you describe your environment, so we can try to reproduce the error?

jschleicher, 2019-03-21:
@ipa-hsd 
Steps to reproduce:
1. `roslaunch prbt_moveit_config moveit_planning_execution.launch sim:=False`
2. Press Ctrl+C to terminate the process
-> Crash reporter pops up

I could get a backtrace starting canopen_motor_node with gdb and sending `kill -s SIGINT` from another terminal to the process (disabling the gdb signal handling as [described here](https://stackoverflow.com/questions/36993909/debugging-a-program-that-uses-sigint-with-gdb)). Please find the backtrace attached.
Commit version for the kinetic backtrace: 98e2b15
[canopen_motor_node-kinetic-segfault-on-shutdown.backtrace.txt](https://github.com/ros-industrial/ros_canopen/files/2992935/canopen_motor_node-kinetic-segfault-on-shutdown.backtrace.txt)

and similar backtrace for melodic package version
```
ii  ros-melodic-canopen-motor-node-dbgsym        0.8.0-0bionic.20181112.190503                amd64
```
[canopen_motor_node-melodic-segfault-on-shutdown.backtrace.txt](https://github.com/ros-industrial/ros_canopen/files/2992936/canopen_motor_node-melodic-segfault-on-shutdown.backtrace.txt)

ipa-mdl, 2019-03-21:
Thanks for the backtraces!
[This](https://github.com/ros-industrial/ros_canopen/blob/162c0a8b20fda2f2a4b13ab1c26e1437beeacdb3/canopen_motor_node/src/handle_layer.cpp#L95) looks suspicious..
I really don't know why `LimitsHandleBase` is not as base class of `LimitsHandle`.

jschleicher, 2019-03-22:
Can you reproduce the issue? What other information do you need?

ipa-hsd-sm, 2019-04-12:
I reproduced and tested the same step you told, in my system, but I did not get any error as such. Below you can see the terminal screen on pressing Control+ C :

**### ^C[rviz_326_stud_2_20224_7350066328424509740-7] killing on exit
[move_group-6] killing on exit
[prbt/controller_spawner-4] killing on exit
[prbt/driver-3] killing on exit
[prbt/joint_state_relay-2] killing on exit
[robot_state_publisher-1] killing on exit
EMCY: 86#5082111000001743
EMCY: 85#5082111000001743
EMCY: 83#5082111000001743
EMCY: 88#5082111000001743
EMCY: 84#5082111000001743
EMCY: 87#5082111000001743
[WARN] [1555077700.654954]: Controller Spawner error while taking down controllers: unable to connect to service: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
EMCY: 83#0000200000000000
EMCY: 84#0000200000000000
EMCY: 85#0000200000000000
EMCY: 86#0000200000000000
EMCY: 87#0000200000000000
EMCY: 88#0000200000000000
shutting down processing monitor...
... shutting down processing monitor complete
done**

Maybe if you can provide us some more reproduction steps, we could figure out the problem.

jschleicher, 2019-04-12:
Do you have the crash reporter installed and activated?
or:
3. check the ros log files, mine looks like (note the return value of prbt/driver aka canopen_motor_node)
```
[roslaunch][INFO]  [robot_state_publisher-1] sending SIGINT to pgid [29318]
[roslaunch][INFO]  [prbt/joint_state_relay-2] sent SIGINT to pgid [29319]
[roslaunch][INFO]  [robot_state_publisher-1] sent SIGINT to pgid [29318]
[roslaunch][INFO]  process[prbt/driver-3]: SIGINT killed with return value -11
[roslaunch][INFO]  process[robot_state_publisher-1]: SIGINT killed with return value 0
[roslaunch][INFO]  process[prbt/joint_state_relay-2]: SIGINT killed with return value 0
[roslaunch][INFO]  process[prbt/controller_spawner-4]: SIGINT killed with return value 0

```

mistoll, 2019-06-19:
Segfault on shutdown happens on my system, too.

Steps:

1. Run canopen_motor_node
2. rosservice call /driver/init
3. Send SIGINT (Ctrl+C)

GDB output:

```
Thread 1 "canopen_motor_n" received signal SIGSEGV, Segmentation fault.
0xb6fd2b90 in typeinfo for LimitsHandle<joint_limits_interface::VelocityJointSoftLimitsHandle> ()
   from /opt/ros/melodic/lib/libcanopen_motor.so
(gdb) bt
#0  0xb6fd2b90 in typeinfo for LimitsHandle<joint_limits_interface::VelocityJointSoftLimitsHandle> ()
   from /opt/ros/melodic/lib/libcanopen_motor.so
#1  0x00408f34 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()
#2  0xb6f9dc78 in canopen::HandleLayer::~HandleLayer() () from /opt/ros/melodic/lib/libcanopen_motor.so
#3  0xb6fa3574 in canopen::LayerGroup<canopen::HandleLayerBase>::~LayerGroup() () from /opt/ros/melodic/lib/libcanopen_motor.so
#4  0xb6fa76b4 in canopen::RobotLayer::~RobotLayer() () from /opt/ros/melodic/lib/libcanopen_motor.so
#5  0xb6fa7708 in canopen::RobotLayer::~RobotLayer() () from /opt/ros/melodic/lib/libcanopen_motor.so
#6  0xb6f806a2 in canopen::ControllerManagerLayer::~ControllerManagerLayer() () from /opt/ros/melodic/lib/libcanopen_motor.so
#7  0xb6ef8890 in canopen::LayerGroup<canopen::Layer>::~LayerGroup() () from /opt/ros/melodic/lib/libcanopen_ros_chain.so
#8  0xb6eeb1fe in canopen::RosChain::~RosChain() () from /opt/ros/melodic/lib/libcanopen_ros_chain.so
#9  0x00408fa8 in canopen::MotorChain::~MotorChain() ()
#10 0x00406bcc in main ()
```

System: armhf, linux 4.14, melodic, ros_canopen 0.8.0

ipa-mdl, 2019-06-19:
I would blame https://github.com/ros-industrial/ros_canopen/blob/162c0a8b20fda2f2a4b13ab1c26e1437beeacdb3/canopen_motor_node/src/handle_layer.cpp#L95

PR #338 (2019-06-24): inherit LimitsHandle from LimitsHandleBase
fixes #324 
@jschleicher, @mistoll: please test
closed 2019-06-24 by ipa-mdl