ros-industrial/ros_canopen #326: rosservice call /arm/driver/init issue

KaushalkumarPatel, 2019-04-01:
Hello to all,

I am using ROS kinetic to init schunk lwa4p arm with esd CAN-USB2. 
After launching the Robot.launch file, When I call rosservice   /arm/driver/init   it give following error.

$ rosservice call /arm/driver/init
success: False
message: "/home/catkin_ws/src/ros_canopen/canopen_master/src/sdo.cpp(429): Throw\
  \ in function void canopen::SDOClient::transmitAndWait(const canopen::ObjectDict::Entry&,\
  \ const canopen::String&, canopen::String*)\nDynamic exception type: boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<canopen::TimeoutException>\
  \ >\nstd::exception::what: SDO\n[canopen::tag_objectdict_key*] = 1401sub1\n"

$ rosservice call /arm/driver/init
success: False
message: "/home/catkin_ws/src/ros_canopen/canopen_master/src/sdo.cpp(429): Throw\
  \ in function void canopen::SDOClient::transmitAndWait(const canopen::ObjectDict::Entry&,\
  \ const canopen::String&, canopen::String*)\nDynamic exception type: boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<canopen::TimeoutException>\
  \ >\nstd::exception::what: SDO\n[canopen::tag_objectdict_key*] = 606bsub0\n"

I am new to ROS. I dont have much idea about this. Can anyone help me with this ?

gavanderhoorn, 2019-04-01:
Also on ROS Answers: [rosservice call /arm/driver/init issue](https://answers.ros.org/question/320053).

ipa-mdl, 2019-04-01:
Closing here in favor of ROS answers.

closed 2019-04-01 by ipa-mdl