ros-industrial/ros_canopen #328: Cannot switch to profile velocity mode after initialization

ishugoel20, 2019-04-25:
Hi,

I am working with engel can motor and after initializing the motor with driver/init service, I cannot switch the mode to profile velocity mode. I get the error that 'mode does not match'. On debugging the can log, I found that the halt bit is set to '1' after the finish of init service which prevents the controller from switching modes. 

If I set the Halt bit to '0' in line 416 of motor.cpp inside the function Motor402::handleWrite, I can switch mode to velocity and provide velocity values. I do not know if this is good fix for this problem? Should I change the halt bit using a parameter? Is there any other way to solve this problem? 

Kindly provide suggestions for the same.

Thanks,
Ishu

ipa-mdl, 2019-04-30:
The Halt bit is crucial for safe operation and control, you should not remove it.
You might use a different switching state (in your config):
```yaml
  motor_layer: settings passed to motor layer (plugin-specific)
    switching_state: 5 # (Operation_Enable), state for mode switching
```
You can find the values in this [enum](https://github.com/ros-industrial/ros_canopen/blob/melodic-devel/canopen_402/include/canopen_402/motor.h#L37).

According to the 402 standard the state is vendor-specific.
You drive's manual might list the correct state.

ishugoel20, 2019-05-10:
@ipa-mdl Thanks for your response. I already tried that solution, it did not work. 

In the [datasheet of the motor at page no. 38](https://www.engelantriebe.de/pdf/CANopen_HB_engl.pdf) , it clearly mentions that "Changing the mode of operation is not allowed, when the HALT-bit (Bit 8 of Controlword6040h, chapter 5.3.1) is set!". 

If I create a variable which can be used to set or reset the Halt bit based on the motor, will it be an acceptable change? Also, is it possible to change the default value of Modes of Operation from 1 (Profile Position Mode) to 3 (Profile Velocity Mode)?

ipa-mdl, 2019-05-10:
> If I create a variable which can be used to set or reset the Halt bit based on the motor, will it be an acceptable change?

I would suggest to reset the Halt bit as long as the motor operation is not enabled and or is not in any mode (#331 ).
Toggling the halt bit in operation-enabled mode is not safe while switching modes.

> Also, is it possible to change the default value of Modes of Operation from 1 (Profile Position Mode) to 3 (Profile Velocity Mode)?

Just set specify the parameter in your DCF/EDS or `dcf_overlay`.

mzahana, 2022-06-01:
@ishugoel20 Hi!
Where you able to change the default mode of operation to Profile Velocity?