ros-industrial/ros_canopen #330: the period for sending CAN frame

jacknlliu, 2019-05-03:
https://github.com/ros-industrial/ros_canopen/blob/e9515ef02b53e899e1b45e4bb48c6d24309fba6a/canopen_chain_node/src/ros_chain.cpp#L86

How about the period about controller manager update if the `sync_` is not false?

I use the `position_controllers/JointPositionController` to send once command via ros command topic, and the PDO are only send to drive three times, but not periodically.

ipa-mdl, 2019-05-03:
> How about the period about controller manager update if the sync_ is not false?

Then `sync_`  will take care of it.

closed 2019-05-03 by ipa-mdl

jacknlliu, 2019-05-03:
....  @ipa-mdl  how does it take care??? Where show the period with `sync`?

ipa-mdl, 2019-05-03:
> how does it take care??? Where show the runtime period with sync?

I don't understand your questions.
If you specify `interval_ms`, a [sync master](http://wiki.ros.org/canopen_master#Master_plug-ins) is used to control the timing (`sync_`).
Without a sync handler, the update interval gets enforced with the code above.

jacknlliu, 2019-05-04:
@ipa-mdl  thanks your tips. You have got it. Is it correct that, if I set the `interval_ms` as 10, then 5ms to `read()`, and 5ms to `write()` SYNC? And then the period of the controller manager update is 10ms.

I got it from here
https://github.com/ros-industrial/ros_canopen/blob/e9515ef02b53e899e1b45e4bb48c6d24309fba6a/canopen_master/src/master_plugin.cpp#L49