ros-industrial/ros_canopen #352: socketcan_bridge package installation

Prj1508, 2019-09-11:
Hi, 

I am trying to use socketcan_bridge package to send CAN commands. I have ROS basic experience and I am new to ros_canopen/socketcan_bridge. 

My socketcan on Ubuntu linux is working fine with a CAN controller and I can read all CAN Messages in the bus as of now. 

I have cloned ros_canopen into my ROS installation and I am trying to build using catkin make, but it is failing with the following error message. Please let me know the resolution

####
#### Running command: "cmake /home/jpm/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/jpm/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/jpm/catkin_ws/install -G Unix Makefiles" in "/home/jpm/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/jpm/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/jpm/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/jpm/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/jpm/catkin_ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/local/bin/nosetests-2.7
-- catkin 0.7.18
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 8 packages in topological order:
-- ~~  - ros_canopen (metapackage)
-- ~~  - socketcan_interface
-- ~~  - canopen_master
-- ~~  - canopen_402
-- ~~  - can_msgs
-- ~~  - canopen_chain_node
-- ~~  - socketcan_bridge
-- ~~  - canopen_motor_node
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'ros_canopen'
-- ==> add_subdirectory(ros_canopen/ros_canopen)
-- +++ processing catkin package: 'socketcan_interface'
-- ==> add_subdirectory(ros_canopen/socketcan_interface)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   chrono
--   system
--   thread
--   date_time
--   atomic
-- +++ processing catkin package: 'canopen_master'
-- ==> add_subdirectory(ros_canopen/canopen_master)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   atomic
--   chrono
--   thread
--   system
--   date_time
-- +++ processing catkin package: 'canopen_402'
-- ==> add_subdirectory(ros_canopen/canopen_402)
-- Boost version: 1.58.0
-- +++ processing catkin package: 'can_msgs'
-- ==> add_subdirectory(ros_canopen/can_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- can_msgs: 1 messages, 0 services
-- +++ processing catkin package: 'canopen_chain_node'
-- ==> add_subdirectory(ros_canopen/canopen_chain_node)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- canopen_chain_node: 0 messages, 2 services
-- +++ processing catkin package: 'socketcan_bridge'
-- ==> add_subdirectory(ros_canopen/socketcan_bridge)
-- +++ processing catkin package: 'canopen_motor_node'
-- ==> add_subdirectory(ros_canopen/canopen_motor_node)
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   chrono
--   system
--   date_time
--   atomic
-- Could NOT find MUPARSER (missing:  MUPARSER_LIBRARY MUPARSER_INCLUDE_DIR) 
CMake Error at ros_canopen/canopen_motor_node/CMakeLists.txt:47 (message):
  muparser library not found

-- Configuring incomplete, errors occurred!
See also "/home/jpm/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/jpm/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

ipa-mdl, 2019-09-13:
As apparent from the error message, you did not install the dependencies. In this case: muparser.

```
Could NOT find MUPARSER (missing: MUPARSER_LIBRARY MUPARSER_INCLUDE_DIR)
CMake Error at ros_canopen/canopen_motor_node/CMakeLists.txt:47 (message):
muparser library not found
```

I would recommend to use `rosdep` to install them.
**This should be your workflow for all cloned ROS repsositories**

As an alternative, you could just install the released Debian package with `apt-get`.

Since this issue is not a bug and this issue tracker is not a support forum, I will close this.

closed 2019-09-13 by ipa-mdl

gavanderhoorn, 2019-09-13:
Related ROS Answers Q&A: [How to build a package from source on Ubuntu MATE 16.04 LTS and ROS Kinetic?](https://answers.ros.org/question/252478) (answer is generic, not tied to 16.04 (or Ubuntu actually).

Prj1508, 2019-09-13:
I installed muparser, but I am getting build errors as below
[ 91%] Building CXX object ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor.dir/src/controller_manager_layer.cpp.o
[ 92%] Building CXX object ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor.dir/src/motor_chain.cpp.o
[ 94%] Building CXX object ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor.dir/src/robot_layer.cpp.o
In file included from /usr/include/c++/5/bits/hashtable.h:35:0,
                 from /usr/include/c++/5/unordered_map:47,
                 from /home/jpm/catkin_ws/src/ros_canopen/canopen_motor_node/include/canopen_motor_node/handle_layer.h:5,
                 from /home/jpm/catkin_ws/src/ros_canopen/canopen_motor_node/src/handle_layer.cpp:2:
/usr/include/c++/5/bits/hashtable_policy.h: In instantiation of ‘struct std::__detail::__is_noexcept_hash<canopen::MotorBase::OperationMode, std::hash<canopen::MotorBase::OperationMode> >’:
/usr/include/c++/5/type_traits:137:12:   required from ‘struct std::__and_<std::__is_fast_hash<std::hash<canopen::MotorBase::OperationMode> >, std::__detail::__is_noexcept_hash<canopen::MotorBase::OperationMode, std::hash<canopen::MotorBase::OperationMode> > >’
/usr/include/c++/5/type_traits:148:38:   required from ‘struct std::__not_<std::__and_<std::__is_fast_hash<std::hash<canopen::MotorBase::OperationMode> >, std::__detail::__is_noexcept_hash<canopen::MotorBase::OperationMode, std::hash<canopen::MotorBase::OperationMode> > > >’
/usr/include/c++/5/bits/unordered_map.h:100:66:   required from ‘class std::unordered_map<canopen::MotorBase::OperationMode, hardware_interface::JointHandle*>’
/home/jpm/catkin_ws/src/ros_canopen/canopen_motor_node/include/canopen_motor_node/handle_layer.h:104:16:   required from here
/usr/include/c++/5/bits/hashtable_policy.h:85:34: error: no match for call to ‘(const std::hash<canopen::MotorBase::OperationMode>) (const canopen::MotorBase::OperationMode&)’
  noexcept(declval<const _Hash&>()(declval<const _Key&>()))>
                                  ^
In file included from /usr/include/c++/5/bits/move.h:57:0,
                 from /usr/include/c++/5/bits/stl_pair.h:59,
                 from /usr/include/c++/5/bits/stl_algobase.h:64,
                 from /usr/include/c++/5/memory:62,
                 from /home/jpm/catkin_ws/src/ros_canopen/canopen_motor_node/include/canopen_motor_node/handle_layer.h:4,
                 from /home/jpm/catkin_ws/src/ros_canopen/canopen_motor_node/src/handle_layer.cpp:2:
/usr/include/c++/5/type_traits: In instantiation of ‘struct std::__not_<std::__and_<std::__is_fast_hash<std::hash<canopen::MotorBase::OperationMode> >, std::__detail::__is_noexcept_hash<canopen::MotorBase::OperationMode, std::hash<canopen::MotorBase::OperationMode> > > >’:
/usr/include/c++/5/bits/unordered_map.h:100:66:   required from ‘class std::unordered_map<canopen::MotorBase::OperationMode, hardware_interface::JointHandle*>’
/home/jpm/catkin_ws/src/ros_canopen/canopen_motor_node/include/canopen_motor_node/handle_layer.h:104:16:   required from here
/usr/include/c++/5/type_traits:148:38: error: ‘value’ is not a member of ‘std::__and_<std::__is_fast_hash<std::hash<canopen::MotorBase::OperationMode> >, std::__detail::__is_noexcept_hash<canopen::MotorBase::OperationMode, std::hash<canopen::MotorBase::OperationMode> > >’
     : public integral_constant<bool, !_Pp::value>
                                      ^
In file included from /usr/include/c++/5/unordered_map:48:0,
                 from /home/jpm/catkin_ws/src/ros_canopen/canopen_motor_node/include/canopen_motor_node/handle_layer.h:5,
                 from /home/jpm/catkin_ws/src/ros_canopen/canopen_motor_node/src/handle_layer.cpp:2:
/usr/include/c++/5/bits/unordered_map.h: In instantiation of ‘class std::unordered_map<canopen::MotorBase::OperationMode, hardware_interface::JointHandle*>’:
/home/jpm/catkin_ws/src/ros_canopen/canopen_motor_node/include/canopen_motor_node/handle_layer.h:104:16:   required from here
/usr/include/c++/5/bits/unordered_map.h:100:66: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<canopen::MotorBase::OperationMode> >, std::__detail::__is_noexcept_hash<canopen::MotorBase::OperationMode, std::hash<canopen::MotorBase::OperationMode> > > >’
       typedef __umap_hashtable<_Key, _Tp, _Hash, _Pred, _Alloc>  _Hashtable;
                                                                  ^
/usr/include/c++/5/bits/unordered_map.h:107:45: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<canopen::MotorBase::OperationMode> >, std::__detail::__is_noexcept_hash<canopen::MotorBase::OperationMode, std::hash<canopen::MotorBase::OperationMode> > > >’
       typedef typename _Hashtable::key_type key_type;
                                             ^
/usr/include/c++/5/bits/unordered_map.h:108:47: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<canopen::MotorBase::OperationMode> >, std::__detail::__is_noexcept_hash<canopen::MotorBase::OperationMode, std::hash<canopen::MotorBase::OperationMode> > > >’
       typedef typename _Hashtable::value_type value_type;
                                               ^
/usr/include/c++/5/bits/unordered_map.h:109:48: error: ‘value’ is not a member of ‘std::__not_<std::__and_<std::__is_fast_hash<std::hash<canopen::MotorBase::OperationMode> >, std::__detail::__is_noexcept_hash<canopen::MotorBase::OperationMode, std::hash<canopen::MotorBa [... cut ...]
ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor.dir/build.make:110: recipe for target 'ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor.dir/src/motor_chain.cpp.o' failed
make[2]: *** [ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor.dir/src/motor_chain.cpp.o] Error 1
CMakeFiles/Makefile2:4930: recipe for target 'ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor.dir/all' failed
make[1]: *** [ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed