ros-industrial/ros_canopen #354: #documentation and instruction how to use ROS socketcan bridge code

duongvuhyperx, 2019-09-25:
Hi,
  Thank you for making a the great plug in, great job and well done!

   I wondering is there any document on how to use the code, like wiki, or the comment of what suppose to go into each line. This will greatly help new developer. 

Similiar wiki: python-can.readthedocs.io/en/master/

Thanks

ipa-mdl, 2019-09-25:
There is not much documentation for developers for `socketcan_bridge` (`ros_canopen`) in general.
However, the code base of `socketcan_bridge` is fairly small..

The API can be seen [here](http://docs.ros.org/melodic/api/socketcan_bridge/html/namespacesocketcan__bridge.html), but without any explanations.

It would be highly appreciated if you would provide comments (doxygen) or explanations in the [wiki](http://wiki.ros.org/socketcan_bridge) as soon as git familiar with the code.

iwanders, 2019-09-25:
When I originally wrote the `socketcan_bridge` I wrote documentation for users (not developers) on the ros wiki. See the [socketcan_bridge](http://wiki.ros.org/socketcan_bridge) page there. The code itself is straightforward and easily readable.

There are some more pages on the wiki like the one on [ros_canopen](http://wiki.ros.org/ros_canopen) and [socketcan_interface](http://wiki.ros.org/socketcan_interface), they provide some overview of how the pieces fit together.