ros-industrial/ros_canopen #355: socketcan_bridge - sent_messages: Decimal conversion

Prj1508, 2019-09-25:
I am trying to publish a message (of type can_msgs/Frame) to sent_message topic in socketcan_bridge. For the id i am providing appropriate hex value ( 0x29F) in this case, but when I echo sent_messages, it is actually converting this hex into decimal (671) and hence I am not able to publish this can message correctly. Can you pleas help?

ipa-mdl, 2019-09-25:
> Can you pleas help?

As I already [told you](https://github.com/ros-industrial/ros_canopen/issues/352#issuecomment-
531140132), this the issue tracker is meant for bug reporting and feature requests.

If you need further support, please ask a question at [ROS answers](https://answers.ros.org), if it was not answered there yet.
I will close this issue here.

> For the id i am providing appropriate hex value ( 0x29F) in this case, but when I echo sent_messages, it is actually converting this hex into decimal (671)

In most (if not all) programming languages, `0x29F` and `671` can be interchanged, because they are just different representations (for humans!) of the same data, which in this case is a 32-bit unsigned integer.

closed 2019-09-25 by ipa-mdl

Prj1508, 2019-09-25:
thanks. I have asked this question on ROS Answers already, but havent got any reponse. 
I know those 2 are same values, but I think the CAN frame should have exact unique id (29F in this case) to be heard by CAN bus.

Prj1508, 2019-09-25:
Also, this hex to decimal conversion happens only when I am on can0. When I am using vcan0, there is no conversion and I can see hex as is when I echo this topic