ros-industrial/ros_canopen #365: ROS Error : Failed to send message

Prj1508, 2019-12-05:
I get this message when I publish on sent_messages topic of socketcan_bridge node. I see that this is getting published properly, but is not being sent to the can bus ( topic to socketcan issue). CAN is working properly and I can see traffic thru candump. 

In topic to socketcan code, I see that there is a boolean flag which sends this message when fault. Is this to do with overflow of data I am trying to stuff into available data bits of the message

I am using Ubuntu 16.04 and ROS Kinetic

thanks for your help

ipa-mdl, 2019-12-09:
I would suspect that your frame is not valid.
How do you publish the data?

Prj1508, 2019-12-09:
i am publishing thru a 10ms timer callback. All CAN data byte stuffing is happening in another subscriber call back which updates a global variable of type can_msgs::Frame. This variable is the. published theu a 10ms timer callback.

Prj1508, 2019-12-09:
this issue is same as #306.

Prj1508, 2019-12-09:
I didn’t not capture console output with —screen option. will do so today and revert

ipa-mdl, 2019-12-09:
> this issue is same as 306.

306 or 366?

Prj1508, 2019-12-09:
Sorry, 366

ipa-mdl, 2019-12-09:
duplicate of #366

closed 2019-12-09 by ipa-mdl