ros-industrial/ros_canopen #366: Error due to driver_state error can::State

Prj1508, 2019-12-06:
I have a publisher in socketcan_bridge that sends message on sent_messages topic. When this is getting converted to socketcan in topic_to_socketcan node, I am getting an error message that says it failed to send the message on to CAN bus. When I debugged, I noticed that can::State is Open which sets a boolean resulting in failure to send CAN message. Here is the snapshot of the debugged state
[image]
To make sure, I sent another message with simpler, but similar data and the message is being transmitted to CAN bus and also can::State is ready. Here is the snapshot
[image]

I am nor sure why my message is not being published to CAN In the first case. Appreciate any suggestions on this problem

ipa-mdl, 2019-12-09:
How is this issue different from #365?

>  I am getting an error message that says it failed to send the message on to CAN bus. 

Which?

>  Here is the snapshot of the debugged state

This does not help much.
Did you see any console output?
If you use `roslaunch´, please make sure to set `--screen` to see all output.

Prj1508, 2019-12-09:
Here is the console output

process[socketcan_bridge_node-1]: started with pid [780]
process[topic_to_socketcan_node-2]: started with pid [781]
process[socketcan_to_topic_node-3]: started with pid [782]
process[DbwRosClass-4]: started with pid [783]

FAILED system:105
[ INFO] [1575928135.658389222]: State: OK, asio: No buffer space available
[ INFO] [1575928135.658453505]: State: OK, asio: No buffer space available

[ERROR] [1575928135.905479353]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.906224001]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.906946850]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.907610312]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.908095260]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.908794152]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.909385881]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.909919792]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.910306181]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.911137633]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.911830380]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.912341156]: Failed to send message: 10#5660000010130000.
[ERROR] [1575928135.912666453]: Failed to send message: 10#5660000010130000.

ipa-mdl, 2019-12-09:
> FAILED system:105
[ INFO] [1575928135.658389222]: State: OK, asio: No buffer space available

The driver stops working because yor are sending messages (way) faster than the underlying system can process it. The errors occur with a rate of 2 KHz.
If it just a burst, you could increate the queue length (https://github.com/ros-industrial/ros_canopen/issues/102).

Anyway, I will close this issue, because it does not seem to be a bug.
If you need support, please consider asking on [ROS answers](http://answers.ros.org/) instead.

> I am not sure why it is trying to send a message with 0# id

I guess something is wrong with your data generation scheme; and it is not thread-safe.

closed 2019-12-09 by ipa-mdl

Prj1508, 2019-12-09:
@ipa-mdl, I did some more analysis. My code is such that I am subscribing to a topic in a callback, then building a message of type can_msgs::Frame in the callback. Now, if I publish from within this callback on sent_messages topic, I see that the message is sent successfully to the bus. 

However, I have to ensure the message should be sent at 10ms rate and hence I have created a timer in ROS (ros::Timer) and then publishing the message I want (which is a global variable in my class definition) from the timercallback. When I do this, the message is not sent and the console output is as above. Please suggest