ros-industrial/ros_canopen #368: Can I modify Absolute Move to Relative Move in position mode?

clm330, 2019-12-16:
By default, ros_canopen set control_word to 0x3F when I pub the position cmd  in position mode, it set the driver to Absolute Move.
Now I find how to modify control_word to 0x5F in source code, but I wonder easier way to modify.
Can somebody help?
Thanks.

ipa-mdl, 2019-12-16:
> By default, ros_canopen set control_word to 0x3F when I pub the position cmd in position mode, it set the driver to Absolute Move.

It is way because absolute commands are the default for `ros_control` as well.

> Now I find how to modify control_word to 0x5F in source code, but I wonder easier way to modify.

There is no easier way.
If you don't want to apply this patch all the time, you can subclass canopen_402 in your custom plugin, which replaces the original profiled position mode with yours.
Or you could extend the existing code as well.
Or you could wire your own controller, which does the integration.

That being said, I would not recommend using relative positions , because the rest of the code assume absolute commands.

clm330, 2019-12-17:
It is helpful.
Thanks.

closed 2019-12-17 by clm330