ros-industrial/ros_canopen #369: Motor jitter when init.

clm330, 2019-12-17:
When drivers init, the motor which specified to position mode jitter.
When init. The mode of operation(0x6060) set to 0, and target position(0x607a) jitter.
It occurs always.

The following is information about the issues. 

**version:kinetic-devel**

This is graph.
[image]

This is config.yaml.
`bus:
  device: can0 # socketcan network
  master_allocator: canopen::SimpleMaster::Allocator # defaults to canopen::LocalMaster::Allocator
sync:
  interval_ms: 200 # set to 0 to disable sync
  overflow: 0 # overflow sync counter at value or do not set it (0, default)

defaults:
  eds_pkg: motor_control
  eds_file: "launch/base/config/DCHCAN.eds"
  pos_to_device: "rint((pos)*1592)" # rad -> mdeg
  pos_from_device: "obj6064*0.000628" 
  vel_to_device: "rint((vel)*15923)"
  vel_from_device: "obj606C*0.0000628" # actual velocity [mdeg/s] -> rad/s
  publish: ["6060","607a","60ff","6098"]
  dcf_overlay: # "ObjectID": "ParameterValue" (both as strings)
    "1017": "100" # heartbeat producer

nodes:
  axis1_to_base_link:
    id: 1
    dcf_overlay:
      "6098": "0" #  homing method
  wheel1_to_axis1:
    id: 2`

This is part of eds.

`[6098]
ParameterName=Homing method
ObjectType=7
DataType=2
AccessType=RW
PDOMapping=1
DefaultValue=0

--------------

[1602]
ParameterName=Receive PDO Mapping Parameter 2
ObjectType=0x9
SubNumber=9

[1602sub0]
ParameterName=Number of entries
ObjectType=0x7
DataType=0x0005
AccessType=rw
PDOMapping=0
LowLimit=0
HighLimit=8
ParameterValue=2

[1602sub1]
ParameterName=PDO Mapping Entry
ObjectType=0x7
DataType=0x0007
AccessType=rw
PDOMapping=0
ParameterValue=0x607a0020

[1602sub2]
ParameterName=PDO Mapping Entry_2
ObjectType=0x7
DataType=0x0007
AccessType=rw
PDOMapping=0
ParameterValue=0x60ff0020`

This is candump information about RPDO3 when init.
[image]

Expecting solution.
Thanks.

ipa-mdl, 2019-12-17:
`0x607a` should not be used (or sent) if the mode is not 1.
What is your position (0x6064)?

Which controller are you using?
Why is your velocity factor 10 times bigger?

clm330, 2019-12-18:
Thanks.

> Which controller are you using?
My controller is `position_controllers/JointPositionController`.

> Why is your velocity factor 10 times bigger?
Because velocity unit is 10 times bigger than position unit.

> What is your position (0x6064)?
In this graph, the mode of operation (0x6060) was set to 1. the homing method(0x6098) was set to 0.
But in begining, the mode of operation display(0x6061) was not same with mode. it was 6(homing mode).
After mode of operation display(0x6061) switch to 1, 0x607A has recevied position message.

[image]

clm330, 2019-12-18:
By using `position_controllers/JointPositionController`, target position is set to 62.8 (rotate 10 rounds). 
In RPDO, the drvier (0x607A) received position Intermittently and Decreasing.
How can send the position driectly?
[image]

clm330, 2019-12-22:
I set 0x607A to 0, and then init normally. But still dont konw why 0x6061 from 1 to 6 when init.

closed 2019-12-22 by clm330

ipa-mdl, 2019-12-22:
> But still dont konw why 0x6061 from 1 to 6 when init.

This is just part of `init`, as it will try to home the device.

clm330, 2019-12-23:
> > But still dont konw why 0x6061 from 1 to 6 when init.
> 
> This is just part of `init`, as it will try to home the device.

Thanks for reply.
Sorry for my careless. I forgot said I also set `0x6098` to 0, the driver always try to home the device. 

My question:
Normally, if `0x6098` set to 0, should `0x6061` be from 1 to 6  when `init`?
Expecting your reply.