ros-industrial/ros_canopen #373: When positon mode, driver run previous position instead of the current position set on 0x607A

clm330, 2019-12-22:
When I sent position command to driver in position mode, the motor always rotate to previous position set on 0x607A.

`ros-canopen version: kinetic`
`controller:joint_state_controller/JointStateController`

Graph:
See the graph. When 0x6040 rising delay, the driver always run to previous position.
[image]

I try to  set a delay at the rising delay to solve this problem but it seems no effect.

cpp file: canopen_402/src/motor.cpp
function: Motor402::handleWrite(LayerStatus &status, const LayerState &current_state)

```
            if(okay) {
                **if(this->mode_id_ == 1){
                    sleep(0.1);
                }**
                std::cout<<"~(1<<Command402::CW_Halt) : "<<std::hex << static_cast<int>(~(1<<Command402::CW_Halt))<<std::endl;
                control_word_ &= ~(1<<Command402::CW_Halt);
                std::cout<<"drive_id :"<<this->name<<" control word :"<<std::hex << static_cast<int>(control_word_)<<std::endl;
            }
```

Maybe I added the delay in the wrong place, can tell me where to add the delay. It may help debug.
Thanks.

ipa-mdl, 2019-12-22:
> See the graph

Are you sure that the motor does not just drive intro the wrong direction?
Could you prepare a graph that steps the position up?

> I try to set a delay at the rising delay

If you add a delay to `handleWrite`, you prolong the communication cycle. These function should never block like this.

clm330, 2019-12-23:
@ipa-mdl 
See this gragp.
It can prove that the driver always goes to the previous position.
[image]

clm330, 2019-12-23:
In velocity mode, I find similar phenomenon.
I set interval_ms to 500.
See the graph.
[image]

The actual velocity ` 0x606C` begen to change from 0 after two time interval_ms, even if set interval_ms to shorter time interval.

Is this problem  same with the above?

clm330, 2020-03-20:
Set the Transmission type to 255, it works normally.

closed 2020-03-20 by clm330