ros-industrial/ros_canopen #374: Does this package support CAN FD ?

CCodie, 2020-01-06:
Hi developers !
Always thanks for your hard work.

I'm now using this package for my vehicle CAN.
And I wonder that I can use this package for CAN_FD network.
(or if I need to modify some code, letting me know how to would be thankful)

I found that Linux_SocketCAN support CAN_FD here - [Linux_socketcan](https://www.kernel.org/doc/Documentation/networking/can.txt).
And it seems code below does work that I can check with `ifconfig` and it's RX bytes.
`sudo ip link set can0 up type can bitrate 500000 dbitrate 2000000 fd on`

Please let me know... Thanks !

ipa-mdl, 2020-01-08:
`socketcan_interface` does not support CAN FD yet.
Apart from adapting the code to use `canfd_frame` and perhaps the `CAN_RAW_FD_FRAMES` option, the frame length is hard-coded to 8 right now.

CCodie, 2020-01-08:
@ipa-mdl , Thank you for letting me know !
I should find another way to use canfd.

ipa-mdl, 2020-01-09:
You could try #375