ros-industrial/ros_canopen #376: Can Not Ready Issue

JadTawil-theonly, 2020-01-13:
Hello,

I am using the SV2D10-C-CE canopen controller from Applied Motion: https://www.applied-motion.com/produc...

After performing the configuration of the motor parameters with their supplied software, i am attempting to control it through ros-canopen. I made sure to provide the EDS that reflects the parameters in the controller; namely, the PDO mappings (communication and mapping params). The following is the error i am getting from the terminal, when the init service is called.

[ INFO] [1578942089.030965037]: Using fixed control period: 0.010000000
[ INFO] [1578942098.869938415]: Initializing XXX
FAILED system:100
[ERROR] [1578942098.905976794]: CAN not ready
Did not receive a response message
state is 255
[ERROR] [1578942109.890241883]: CAN not ready; not operational; CAN not ready

And on the service client side, i receive:

success: False
message: "could not reset node '1'"

The part that says "state is 255", i am printing myself by adding the following to line 153 in node.cpp

std::cout << "state is " << state_ << std::endl;

I see that state 255 corresponds to "unknown", which is concerning, and suggests that the communication between the controller and the ros-canopen is somehow going wrong. A few details about the controller i am using:

    SV200 servo drivesare compliant to CiA 301 and CiA 402 and uses the CAN 2.0B passive physical layer.

Is the ros-canopen driver programmed to support CAN 2.0B physical layer? I am aware that 2.0B protocol uses 29 bit CAN-ids, as opposed to the 11 bit CAN-ids that protocol 2.0A uses.

Can this be the problem?

Id like to add, i have been using this driver for years with CanOpen controllers that use the 2.0A protocol, which is leading to suspect that this may be the issue here

JadTawil-theonly, 2020-01-13:
Hello,

I realized i had not brought up the Can Interface on my ubuntu computer. Now i am getting a different error:

`[ INFO] [1578945027.787051778]: Using fixed control period: 0.010000000
[ INFO] [1578945030.443453268]: Initializing...
[ INFO] [1578945030.444045328]: Current state: 1 device error: system:0 internal_error: 0 (OK)
[ INFO] [1578945030.444491229]: Current state: 2 device error: system:0 internal_error: 0 (OK)
[ INFO] [1578945041.159381236]: Current state: 2 device error: system:125 internal_error: 0 (OK)
[ INFO] [1578945041.159566863]: Current state: 0 device error: system:125 internal_error: 0 (OK)
[ INFO] [1578945041.159716774]: Current state: 0 device error: system:0 internal_error: 0 (OK)
[ INFO] [1578945041.159911293]: Current state: 0 device error: system:0 internal_error: 0 (OK)
[ERROR] [1578945041.160490380]: Initializing failed: Transition timeout; Could not enable motor; Transition timeout
`

The following text file contains the contents of "candump can0":
https://drive.google.com/open?id=1_8BhBX1DEYX754bl5KkWXsFsBdKNMLUQ

JadBatmobile, 2020-01-13:
What is confusing about the candump is this part:

```
 can0  601   [8]  40 61 60 00 00 00 00 00
 can0  581   [8]  4F 61 60 00 00 00 00 00
```

Which means that there is a SDO read request for object 6061 (Operation mode display), and the response is 00, which makes no sense since 0 is not a valid mode; the modes are:

```
1 Profile Position Mode
3 Profile Velocity Mode
4 Torque Profile Mode (Servo only)
6 Homing Mode
```

Doesn't this driver set the mode of operation on initialization? Perhaps it is hanging because it is getting a non-sense response for modes of operation.

gavanderhoorn, 2020-01-13:
Cross posted to ROS Answers: [Can Not Ready Error](https://answers.ros.org/question/341641)

JadTawil-theonly, 2020-01-13:
Is cross posting frowned on? :/

gavanderhoorn, 2020-01-13:
Yes. Is that a question?

JadTawil-theonly, 2020-01-13:
Okay, will remove the ros answers post

JadTawil-theonly, 2020-01-13:
@gavanderhoorn won't happen again!

closed 2020-01-16 by JadTawil-theonly