ros-industrial/ros_canopen #377: Control word read/write

JadTawil-theonly, 2020-01-16:
Does Ros-canopen assume that the control word (0x6040) is read/write access.
Is it compatible with firmware in which the control word is write-only ?

ipa-mdl, 2020-01-16:
> Does Ros-canopen assume that the control word (0x6040) is read/write access.

Write access should be sufficient, as long as a default is set in the EDS file.
If no default is set, the current value will be read at initialization, because `canopen_402` only changes certain bits.

> firmware in which the control word is write-only

I doubt that this is DSP 402-compliant..

JadTawil-theonly, 2020-01-16:
So you are saying that canopen_402 will read the default, and update the control word internally based on all the changes it makes to it, but would never read it directly from the controller, in the case it is write-only.

I am using the following: https://www.applied-motion.com/sites/default/files/hardware-manuals/CANopen_Manual_920-0025L.pdf

In which case the control word is wo..

ipa-mdl, 2020-01-16:
Normally, it reads the actual value at initialization.
This way previously set bit are kept as-is.
If the default does not match the current value, this might lead to unintended effects.

ipa-mdl, 2020-01-17:
> I doubt that this is DSP 402-compliant..

I just double checked,  `0x6040` is supposed to have acces type `rw` or `rww`.

> So If the control word is write only, the program gracefully proceeds at initialization? 

Only, if a default is set.
If not, it will probably crash with an AccessException. 

> whether it can have something to do with this incorrect implementation detail of wo control word

Did you check if the object is writable?
Sometimes the EDS is wrong..

closed 2021-10-14 by ipa-mdl