ros-industrial/ros_canopen #380: canopen_402 fails with joint_position_controller

JohannLange, 2020-02-21:
Hello,

I'm using `ros_canopen` to controll a motor driver, [C5-E](https://us.nanotec.com/products/1768-c5-e-2-09-motor-controllerdrive-for-canopen-or-usb/) from Nanotec, running on ROS kinetic.

I'm launching my code via `roslaunch myrobot joint_y.launch` and initializing the driver via `rosservice call /can/joint_y/driver/init`. After initializing, I get an error message in the ROS log which reads `Mode handler has error`. I can trace it back to [canopen_402 L388](https://github.com/ros-industrial/ros_canopen/blob/kinetic-devel/canopen_402/src/motor.cpp#L388). However, it's not really descriptive of why it went wrong in the first place.

The error only occurs when I'm using the joint_position_controller. Using the joint_velocity_controller, everything runs just fine.

This is my configuration:

- [joint_y.launch](#joint_y.launch)
  - [joint.xml](#joint.xml)
    - [can_conf.yaml](#can_conf.yaml)
    - [can_joint_y.yaml](#can_joint_y.yaml)
    - [joint_y_limits.yaml](#joint_y_limits.yaml)
    - [joint_y_controller.yaml](#joint_y_controller.yaml)
    - [myrobot.urdf.xacro](#myrobot.urdf.xacro)

With the corresponding logs:

- [ROS log](#ros-log)
- [candump](#candump)

### `joint_y.launch`

```xml
<?xml version="1.0"?>
<launch>
    <arg name="name" value="joint_y" />

    <include file="$(find myrobot)/launch/can/joint.xml">
        <arg name="name" value="$(arg name)" />
    </include>

</launch>
```

### `joint.xml`

```xml
<?xml version="1.0"?>
<launch>
    <arg name="name"/>
    <arg name="timeout" default="180" /> <!-- Timeout (s) for the controller spawners -->

    <!-- drivers -->
    <node ns="$(arg name)" name="driver" pkg="canopen_motor_node" type="canopen_motor_node" output="screen" clear_params="true" launch-prefix="">
        <rosparam command="load" file="$(find myrobot)/cfg/yaml/can/can_conf.yaml" />
        <rosparam command="load" file="$(find myrobot)/cfg/yaml/can/can_$(arg name).yaml" />
        <rosparam command="load" file="$(find myrobot)/cfg/yaml/can/$(arg name)_limits.yaml" />
        <remap from="joint_states" to="/joint_states"/>
    </node>

    <!-- controllers -->
    <rosparam ns="$(arg name)" command="load" file="$(find myrobot)/cfg/yaml/can/$(arg name)_controller.yaml" />

    <node ns="$(arg name)" name="$(arg name)_position_controller_spawner" pkg="controller_manager" type="spawner" args="joint_position_controller --timeout $(arg timeout)" />
    <node ns="$(arg name)" name="$(arg name)_velocity_controller_spawner" pkg="controller_manager" type="spawner" args="--stopped joint_velocity_controller --timeout $(arg timeout)" />
    <node ns="$(arg name)" name="$(arg name)_state_controller_spawner" pkg="controller_manager" type="spawner" args="joint_state_controller" />

    <!-- send urdf to param server -->
    <group ns="$(arg name)">
        <param name="robot_description" command="xacro --inorder '$(find myrobot)/cfg/urdf/myrobot.urdf.xacro'" />
    </group>

    <!-- upload script server parameters -->
    <rosparam command="load" ns="$(arg name)" file="$(find myrobot)/cfg/yaml/can/$(arg name)_configurations.yaml"/>

    <!-- robot state publisher -->
    <node pkg="robot_state_publisher" type="robot_state_publisher" name="robot_state_publisher">
[... 2339 lines cut ...]
  can0  589   [8]  43 6C 60 00 00 00 00 00
  can0  080   [0]
  can0  609   [8]  40 77 60 00 00 00 00 00
  can0  589   [8]  4B 77 60 00 00 00 00 00
  can0  609   [8]  40 6C 60 00 00 00 00 00
  can0  589   [8]  43 6C 60 00 00 00 00 00
  can0  080   [0]
  can0  609   [8]  40 77 60 00 00 00 00 00
  can0  589   [8]  4B 77 60 00 00 00 00 00
  can0  609   [8]  40 6C 60 00 00 00 00 00
  can0  589   [8]  43 6C 60 00 00 00 00 00
  can0  709   [1]  05
  can0  080   [0]
  can0  609   [8]  40 77 60 00 00 00 00 00
  can0  589   [8]  4B 77 60 00 00 00 00 00
  can0  609   [8]  40 6C 60 00 00 00 00 00
  can0  589   [8]  43 6C 60 00 00 00 00 00
  can0  080   [0]
  can0  609   [8]  40 77 60 00 00 00 00 00
  can0  589   [8]  4B 77 60 00 00 00 00 00
  can0  609   [8]  2B 17 10 00 00 00 00 00
  can0  589   [8]  60 17 10 00 00 00 00 00
  can0  000   [2]  02 09
  can0  709   [1]  04
```

ipa-mdl, 2020-02-26:
It looks like you did not map the PDOs: http://wiki.ros.org/canopen_402#Overview
SDO access

JohannLange, 2020-02-27:
Thank you for the insight!
Mind telling me where I would need to add the PDO mapping and how to do it?

BenArtes, 2020-03-13:
Essentially you need to do it manually. 

Set the relevant PDO Parameter / Mapping registers using the `dcf_overlay` key. The dcf_overlay is a way to set values of Object Dictionary Entries at initialization. ros_canopen can use the information in the dcf_overlay to figure out what has been mapped into the PDO so that it uses PDOs instead of SDOs.

A quick example is below:
```yaml

defaults:
  # ...
  dcf_overlay: # This overwrites entries in the EDS / DCF, using it to set acceleration, PIDs, and PDOs
    # RPDO Mappings - From ros_canopen
    "1600sub0": "2" # RPDO1 Mapping Num Entries
    "1400sub2": "1" # RPDO1 Transmission Type
    "1600sub1": "0x60400010" # map Control Word to RPDO1
    "1600sub2": "0x607a0020" # map Target Position to RPDO1
    # TPDO Mappings - To ros_canopen
    "1a00sub0": "3" # TPDO1 Mapping Num Entries
    "1800sub2": "1" # TPDO1 Transmission Type
    "1a00sub1": "0x60410010" # map Status Word to TPDO1
    "1a00sub2": "0x60630020" # map Actual Position to TPDO1
    "1a00sub3": "0x60780010" # map Actual Current to TPDO1
```

JohannLange, 2020-12-18:
Forgot to close the issue.

Basically, I went with [KITmedical/kacanopen](https://github.com/KITmedical/kacanopen), as it worked right out of the box.

I think the problem is somehow related to my CAN setup, especially the CAN chip on my master. For some reason, it was way more robust with `kacanopen` than with `ros_canopen`.

closed 2020-12-18 by JohannLange