ros-industrial/ros_canopen #382: Using CAN bus ID's

BenJamin-ad, 2020-02-25:
I would like to filter the CAN bus messages to only read a few CAN ID's from my vehicle before passing them onto other parts of my system or into a bag as I don't require the entire vehicle CAN bus.

I see that this functionality has been mentioned in #230 #250 but I am struggling to see where I can include my CAN ID's.

I am currently using socketcan_bridge to bring CAN bus data into ROS.

Is it possible for someone to explain how I can only receive messages from a few CAN ID's?

Many thanks.

ipa-mdl, 2020-02-26:
> I see that this functionality has been mentioned in #230 #250 but I am struggling to see where I can include my CAN ID's.

As described in #250, you need to set the `can_ids` parameter (list of numbers and/or strings).

BenJamin-ad, 2020-02-26:
Thank you for the reply.

I appreciate that #250 shows me the parameter to set and the accepted format however I am still confused how I can do this?  Do I set this in the ROS workspace or elsewhere?

ipa-mdl, 2020-02-26:
> however I am still confused how I can do this? Do I set this in the ROS workspace or elsewhere?

A common approach is to create a [launch file](http://wiki.ros.org/roslaunch/XML) that [loads the config from a YAML file](http://wiki.ros.org/roslaunch/XML/rosparam) or [sets the parameters](http://wiki.ros.org/roslaunch/XML/param) directly.

Please note that `can_ids` is a private parameter.

If you need help in setting a parameter, please check [ROS answers](https://answers.ros.org/) as well.

PS: I will close this issue now, because it this is not a ros_canopen-specific issue.

closed 2020-02-26 by ipa-mdl

BenJamin-ad, 2020-02-27:
Thank you @ipa-mdl.

I know this is probably trivial for most but in case anyone is interested here is my launch file that successfully logged only can id 176:

`
 <launch>
<node pkg="socketcan_bridge" type="socketcan_to_topic_node" name="CAN_node" output="screen">
<rosparam param="can_ids">[176]</rosparam>
</node>
</launch>
`

OmerKonan, 2022-04-12:
Hello, I am trying to initialize XmlRpc::XmlRpcValue by hard-coded. I don't want to get value from ros params. How can I create XmlRpc::XmlRpcValue from std::string list. @BenJamin-ad