ros-industrial/ros_canopen #383: send position command in the PVT mode

jackstarred, 2020-03-22:
I find that if we set the drive mode as the PVT mode, the ros canopen only sends the position command to the drive not including the velocity and segment time. Is it right?

ipa-mdl, 2020-03-22:
> set the drive mode as the PVT mode

How do you set PVT mode?
(I recall that I saw "PVT" in some manual, but forgot where..)

> ros canopen only sends the position command to the drive not including the velocity and segment time. Is it right?

The default modes only support sending one command value at a time, only position, velocity or torque.

jackstarred, 2020-03-23:
@ipa-mdl  PVT mode is the IP(Interpolation position) mode, its example can be found at [AMC_CommManual_CANopen](https://info.a-m-c.com/acton/attachment/33817/f-003b/0/-/-/-/-/AMC_CommManual_CANopen.pdf).

I think the appropriate ros control implement for CANopen is the joint position velocity hardware interface, not the joint position hardware interface.