ros-industrial/ros_canopen #399: catkin_make error when using socketcan_interface to build robotiq package

Yitian-Xian, 2020-07-31:
Hi, I am trying to attach a robotiq 2-finger gripper on the ur5 arm robot in ROS. I downloaded the package from https://github.com/ros-industrial/robotiq and I am using ubuntu 18.04 and ROS melodic version. To bulid robotiq package downloaded from the link mentioned, we would need socketcan_interface package. However, when I tried to catkin_make this package in my workspace, it seems that there is one node that could not be bulit, and the error is as follows:

`CMakeFiles/robotiq_3f_gripper_can_node.dir/src/robotiq_3f_gripper_can_node.cpp.o: In function `boost::date_time::counted_time_system<boost::date_time::counted_time_rep<boost::posix_time::millisec_posix_time_system_config> >::subtract_times(boost::date_time::counted_time_rep<boost::posix_time::millisec_posix_time_system_config> const&, boost::date_time::counted_time_rep<boost::posix_time::millisec_posix_time_system_config> const&)':
/home/xian/ros_ws/src/ros_canopen-melodic-devel/socketcan_interface/include/socketcan_interface/socketcan.h:24: multiple definition of `can::parse_error_mask(std::shared_ptr<can::Settings const>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)'
CMakeFiles/robotiq_3f_gripper_can_node.dir/src/robotiq_3f_gripper_control/robotiq_3f_gripper_can_client.cpp.o:/home/xian/ros_ws/src/ros_canopen-melodic-devel/socketcan_interface/include/socketcan_interface/socketcan.h:24: first defined here
collect2: error: ld returned 1 exit status
robotiq/robotiq_3f_gripper_control/CMakeFiles/robotiq_3f_gripper_can_node.dir/build.make:152: recipe for target '/home/xian/ros_ws/devel/lib/robotiq_3f_gripper_control/robotiq_3f_gripper_can_node' failed
make[2]: *** [/home/xian/ros_ws/devel/lib/robotiq_3f_gripper_control/robotiq_3f_gripper_can_node] Error 1
CMakeFiles/Makefile2:2471: recipe for target 'robotiq/robotiq_3f_gripper_control/CMakeFiles/robotiq_3f_gripper_can_node.dir/all' failed
make[1]: *** [robotiq/robotiq_3f_gripper_control/CMakeFiles/robotiq_3f_gripper_can_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

From the result we can know that there are multiple definition of something in socketcan_interface package, but I do not have any idea on how to fix this problem. Can anyone help me? Thanks!

ipa-mdl, 2020-07-31:
Thanks for reporting!

I looked into it and can confirm the problem.
It is more a problem with the robotiq driver, which includes `socketcan,h` instead of `interface.h`.
In a way it is a regression of #362, I will try to rewrite it..

ipa-mdl, 2020-07-31:
@Yitian-Xian:please try again with #400

Yitian-Xian, 2020-07-31:
Hi @ipa-mdl, thanks for your revision!

However, after I revised the h file, I still met this problem. Could you give a check? Thanks!

`CMakeFiles/robotiq_3f_gripper_can_node.dir/src/robotiq_3f_gripper_can_node.cpp.o: In function `can::parse_error_mask(std::shared_ptr<can::Settings const>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)':
robotiq_3f_gripper_can_node.cpp:(.text+0x121): multiple definition of `can::parse_error_mask(std::shared_ptr<can::Settings const>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)'
CMakeFiles/robotiq_3f_gripper_can_node.dir/src/robotiq_3f_gripper_control/robotiq_3f_gripper_can_client.cpp.o:robotiq_3f_gripper_can_client.cpp:(.text+0xa6): first defined here
collect2: error: ld returned 1 exit status
robotiq/robotiq_3f_gripper_control/CMakeFiles/robotiq_3f_gripper_can_node.dir/build.make:152: recipe for target '/home/xian/ros_ws/devel/lib/robotiq_3f_gripper_control/robotiq_3f_gripper_can_node' failed
make[2]: *** [/home/xian/ros_ws/devel/lib/robotiq_3f_gripper_control/robotiq_3f_gripper_can_node] Error 1
CMakeFiles/Makefile2:11920: recipe for target 'robotiq/robotiq_3f_gripper_control/CMakeFiles/robotiq_3f_gripper_can_node.dir/all' failed
make[1]: *** [robotiq/robotiq_3f_gripper_control/CMakeFiles/robotiq_3f_gripper_can_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

ipa-mdl, 2020-07-31:
You might have to clean the build (delete the build directory) first.

Yitian-Xian, 2020-07-31:
Hi @ipa-mdl , I followed your advice and it works well! Thank you!

ipa-mdl, 2020-07-31:
fixed in #400

closed 2020-07-31 by ipa-mdl