ros-industrial/ros_canopen #406: Segmentation fault when using ThreadedSocketCANInterface

ozzdemir, 2020-09-22:
I was using ThreadedSocketCANInterface for interfacing with the CAN network. But with latest update(0.8.4), I am having a segmentation fault. Following is the gdb output. I guess it is related with the change **implemented report-only and fatal errors for SocketCANInterface**. My guess is that it is only implemented for regular socketcan_interface and the situation broke the threaded one.
```
Thread 1 "node" received signal SIGSEGV, Segmentation fault.
can::Settings::get_optional<bool> (this=0x0, n="fatal_error_mask/CAN_ERR_LOSTARB", 
    def=@0x7fffffffc64f: false)
    at /opt/ros/melodic/include/socketcan_interface/settings.h:16
16                if(!getRepr(n, repr)){
```

OS: Ubuntu 18.04
Distro: Melodic

stribor14, 2020-09-22:
+1
same thing happens to us

ipa-mdl, 2020-09-22:
Did you try to clean and rebuild your workspace?

ozzdemir, 2020-09-22:
Just tried cleaning the workspace, and the same result.

ipa-mdl, 2020-09-22:
> Just tried cleaning the workspace, and the same result.

Can you share the code with us?
At least a snippet?

ipa-mdl, 2020-09-22:
I think I found the issue, please try https://github.com/ros-industrial/ros_canopen/pull/407

If you cannot wait for the release, pass the settings explicitly.

ozzdemir, 2020-09-22:
With #407, the problem is solved.

closed 2020-09-22 by ozzdemir

ipa-mdl, 2020-09-22:
Thanks for testing!
I will release it ASAP.

ipa-mdl, 2020-09-22:
https://github.com/ros/rosdistro/pull/26653
https://github.com/ros/rosdistro/pull/26654

PR #407 (2020-09-22): initalize settings properly in deprecated SocketCANInterface::init
Fixes #406