ros-industrial/ros_canopen #409: Not compatible with CombinedRobotHW

ivan140, 2020-09-29:
Hi,

When trying to follow this [CombinedRobotHW](https://github.com/ros-controls/ros_control/wiki/Writing-CombinedRobotHW) tutorial I discovered, that ros_canopen is not capable of supporting the necessary class loader plugin. Are there any plans on adding this capability?

The files in question are: 

- ros_canopen/canopen_motor_node/include/canopen_motor_node/robot_layer.h 

- ros_canopen/canopen_motor_node/include/canopen_motor_node/robot_layer.cpp

The big advantage of this would be the possibility to add custom CAN devices or even Serial controlled devices to a robot running on the CANopen stack.

ipa-mdl, 2020-09-29:
> Are there any plans on adding this capability?

No, this it not planned. For now  `canopen_motor_node` manages the `ros_control` and not the other way around.

bxc237, 2021-07-05:
Hi @ivan140, I am encountering this issue as well, did you find a way to solve it?

ivan140, 2021-07-06:
@bxc237 I luckily have just a passive rotating joint. To solve the upper mentioned issue I added a simple publisher, which publishes the encoder values to the `/joint_states` topic. This way I now have two different publishers:
CANopen ros_control:
`[wheel_left, wheel_right]`

Custom encoder node:
`[body_joint]`

It seems the `robot_state_publisher` is handling this setup just fine, so for me it was a good workaround.

ipa-mdl, 2021-07-06:
@ivan140: If you want to improve your setup even further, add a [joint_state_publisher](http://wiki.ros.org/joint_state_publisher) in between. This will reduce the load of `robot_state_publisher` and all other nodes, which consume `/joint_states` or `/tf`. This is especially useful for MoveIt, which runs collisions checks on each update.