ros-industrial/ros_canopen #410: Services Not Quick Enough For Loop

connoranderson, 2020-11-03:
I'm currently using canopen_chain_node to interface with some motor controllers (not using 402 pkg yet). Before this, I had a basic prototype working with the bridge node connected to my controller, which published the RPDO messages directly to the controller.

Now that I am using the chain node, my question is what are the best ways to achieve low-latency control loops using this package (in a production setting)? It seems like there are a few options:

1. publish to socketcan bridge node /sent_messages
2. canopen_chain_node /driver/set_object method 
3. use socketcan_interface library directly to send message

I've tried mocking out the /set_object implementation, but I'm finding that the execution time is around 8-9ms per service call & SDO response. This isn't acceptable for our application, so I'm curious how others are achieving decent loop rates with that amount of blocking routine?

Can someone with experience on this shed some light on the best-practice here? I realize a lot of people are probably relying on ros_control, but I appreciate any insight into this.

ipa-mdl, 2020-11-03:
> Can someone with experience on this shed some light on the best-practice here?

In general, I would recommend:
4. Subclass `canopen::RosChain` (from `canopen_chain_node`) and `canopen::Node` (from `canopen_master`) to support your device properly, because this gives you full control.

> I've tried mocking out the /set_object implementation, but I'm finding that the execution time is around 8-9ms per service call & SDO response. 

`set_object` is mostly meant for debugging and bound to the update rate (= the call does not get process faster).
Please keep in mind that non-persistent service clients have to establish the connection first.
In additon, SDOs are very slow for some devices. To use PDOs instead, just map the object.

connoranderson, 2020-11-03:
@ipa-mdl thank you for the speedy and detailed response.

So, in general, most people are building wrapper nodes around these classes for their own implementation? Are there any ways to do this through ROS interfaces (for instance if the calling function is in python)?

connoranderson, 2020-11-05:
Going to mark this as closed since the initial question was answered. If anyone has answers to the follow-on questions, it's much appreciated.

closed 2020-11-05 by connoranderson