ros-industrial/ros_canopen #413: Bug? Still sending velocity commands when /driver/halt is called

bxc237, 2020-12-04:
Hello, I map velocity commands 60ff to PDO. When I call /driver/halt, candump still showing those PDOs. Is this a bug?
[image]
I am mapping 60ff to 1602sub1 and controlling two motos with ID 1 and 2 so that the PDO are 401 and 402 here.

ipa-mdl, 2020-12-04:
> Hello, I map velocity commands 60ff to PDO. When I call /driver/halt, candump still showing those PDOs. Is this a bug?

No, this expected, because the input does not change and you seem to use synchronous PDOs.
Please note that the Halt bit will be set as well, so the motor must not move.

The controller (ros_control) might reset the command during the `recover` service.

bxc237, 2020-12-18:
> >
> Please note that the Halt bit will be set as well, so the motor must not move.

Thank you for your reply, then the problem is that motors are still moving after I call `halt` service if these PDOs are still there and those values are not zero...very strange...It seems like the Halt bit does not work. 

Another question is I map 6079 (for the motor driver I am using, is the input voltage) to PDO and when I publish 6079 and monitor the value by subscribing it, the value does not match the value I use candump. The candump value is correct but topic value is wrong. These are my [eds](https://www.dropbox.com/s/tstxc59v9ojsb5d/LS_wheel.eds?dl=0) and [canopen_config.yaml](https://www.dropbox.com/s/dz9m2ywzw65varj/canopen_config.yaml?dl=0) files. Thank you again.

bxc237, 2021-01-11:
> > Hello, I map velocity commands 60ff to PDO. When I call /driver/halt, candump still showing those PDOs. Is this a bug?
> 
> No, this is expected, because the input does not change and you seem to use synchronous PDOs.
> Please note that the Halt bit will be set as well, so the motor must not move.
> 
> The controller (ros_control) might reset the command during the `recover` service.

One more question, how to monitor if the Halt bit was set or not?

ipa-mdl, 2021-01-11:
> One more question, how to monitor if the Halt bit was set or not?

It is the [ninth bit](https://github.com/ros-industrial/ros_canopen/blob/melodic-devel/canopen_402/include/canopen_402/motor.h#L94) of the control word.

bxc237, 2021-02-03:
> > One more question, how to monitor if the Halt bit was set or not?
> 
> It is the [ninth bit](https://github.com/ros-industrial/ros_canopen/blob/melodic-devel/canopen_402/include/canopen_402/motor.h#L94) of the control word.

It seems that the driver is not strictly following the DS301 standard. Close issue!

closed 2021-02-03 by bxc237

ipa-mdl, 2021-02-03:
>It seems that the driver is not strictly following the DS301 standard. Close issue!

Could you explain this a little bit more, please?