ros-industrial/ros_canopen #416: Documentation + support for inclinometers using ros_canopen

fmg-shehan, 2020-12-17:
Hi everyone,
I'm looking for clear documentation on using ros-canopen and would like to know if anyone has any links. 

And I'm also trying to understand if this supports inclinometers using DS410 profile?

ATTN: @ipa-mdl (Seeing as you've contributed  to this repo recently)

Thanks everyone!

BenArtes, 2020-12-21:
At the profile level it implements only DS402 Motor Control. For debug purposes arbitrary Object Dictionary entries can be read from and published on topics and written to using [canopen_chain_node](http://wiki.ros.org/canopen_chain_node?distro=noetic). This is very straightforward if you're trying to access custom ODEs in motor control nodes as you just need to make sure the EDS describes the ODEs and then edit the can_config.yaml rosparams.

If you need non-motor simple CANopen support without any Motors (Just using the debug messaging / topics from canopen_chain_node), it is still pretty straight forward because you can basically just launch a `canopen_chain_node` instead of a `canopen_motor_node`.

If you need non-motor simple CANopen support WITH Motors, you'll need to use multiple processes and external sync as @ipa-mdl referenced in my original question about [Generic Profile support](https://github.com/ros-industrial/ros_canopen/issues/393).

If you need complex profile support, more akin to how motors are integrated with `ros_control` you'll need to extend ros_canopen using the `pluginlib` support yourself. I haven't found any examples of this, nor have I implemented it myself, but once you're reasonably familiar with the ros_canopen source, it should look straightforward; try to replicate canopen_motor_node for your own profile.

fmg-shehan, 2021-08-03:
Hi @BenArtes thanks for your reply. This answered part of my question and I look into the material in more detail.

closed 2021-08-03 by fmg-shehan