ros-industrial/ros_canopen #439: How to skip entering homing mode when I call driver/init service of canopen_motor_node(canopen_chain_node)?

ye-luo-xi-tui, 2021-07-07:
Hi,I find that the `driver/init` would try to make my device enter homing mode.If it failed,init failed too.But manufacturer didn't provide homing method which I need.And I want to make my device home in nonnormal method.So how can I skip the step that enter homing mode of the service `driver/init`?

ipa-mdl, 2021-07-07:
You have 3 options:

* set homing method (`0x6098`) to 0, make it the `DefaultValue` in EDS
* remove the homing mode from supported drive mode `0x6502`
* implement your own homing ([example](https://github.com/ipa320/cob_driver/blob/kinetic_dev/cob_elmo_homing/))

ye-luo-xi-tui, 2021-07-07:
> You have 3 options:
> 
> * set homing method (`0x6098`) to 0, make it the `DefaultValue` in EDS
> * remove the homing mode from supported drive mode `0x6502`
> * implement your own homing ([example](https://github.com/ipa320/cob_driver/blob/kinetic_dev/cob_elmo_homing/))

I will try, thanks.

closed 2021-12-25 by ye-luo-xi-tui