ros-industrial/ros_canopen #443: EffortJointInterface not found when loading joint_effort_controller

ivan140, 2021-09-02:
Hi,

I'm trying to get running the joint_effort_controller with the ros_canopen package.
My motor supports the target_torque obj6071 and it moves when I set some values in this object directly. However, when I try to load the effort_controller I get the following error:

`[INFO] [1630575428.170701]: Loading controller: wheel_left_joint_effort_controller
[ERROR] [1630575428.177604104]: Exception thrown while initializing controller 'wheel_left_joint_effort_controller'.
Could not find resource 'wheel_left_joint' in 'hardware_interface::EffortJointInterface'.
[ERROR] [1630575428.177679782]: Initializing controller 'wheel_left_joint_effort_controller' failed
`
It seems, that the EffortJointInterface is not available. Here are my configs:

**controller.yaml:**
```yaml
joint_names: [wheel_left_joint, wheel_right_joint, body_joint]

## control_mode_adapter
max_command_silence: 0.5

## joint_state_controller
joint_state_controller:
    type: joint_state_controller/JointStateController
    publish_rate: 50

wheel_left_joint_effort_controller:
  type: effort_controllers/JointEffortController
  joint: wheel_left_joint
  required_drive_mode: 4
```
**driver.yaml:**
```yaml
defaults:
  eds_pkg: robot_model
  eds_file: "diff_drive/config/KINCO-JD_FD.eds"
  motor_allocator: canopen::Motor402::Allocator # select allocator for motor layer
  motor_layer: # settings passed to motor layer (plugin-specific)
    switching_state: 5 # (Operation_Enable), state for mode switching
  pos_to_device: "rint(pos*140000/2/pi)" # rad -> encoder ticks
  pos_from_device: "obj6063*2*pi/140000" # actual position [encoder ticks] -> rad
  vel_to_device: "rint(vel*30/pi/1875*512*140000)" # rad/s -> mdeg/s
  vel_from_device: "(obj606C*1875/512/140000)*pi/30" # actual velocity [mdeg/s] -> rad/s
  eff_to_device: "rint(obj6071)" # just round to integer
  eff_from_device: "0" # unset

  dcf_overlay:
    "6007": "0x0001"
    "6081": "250000"
    "6083": "15000"
    "6084": "15000"
    "1016sub1" : "0x7F0064" # heartbeat timeout of 100 ms for master at 127

nodes:
  wheel_left_joint:
    id: 1538
  wheel_right_joint:
    id: 1537
```
Any help would be appreciated!

ipa-mdl, 2021-09-02:
This issue tracker is not a support forum, so please use [ROS answers](https://answers.ros.org) instead.

> My motor supports the target_torque obj6071

`canopen_motor_node` does not check the objects, it just checks the supported [drive modes](http://wiki.ros.org/canopen_402#Drive_operation_modes) from EDS/DCF or directly from the drive.

>   eff_to_device: "rint(obj6071)" # just round to integer

Where did you copy this from?

This would just command the last torque (as read from the device / `6071`).

The effort command (from ros_control) can be read from the `eff` variable and should be passed _to the device_.

closed 2021-09-02 by ipa-mdl

ivan140, 2021-09-02:
Hi,

The error message sounded to me like the effort interface is not implemented thus not loading. That's why I assumed it's an issue.
Also [this table](http://wiki.ros.org/canopen_402#Tested_devices) states, that torque control is not supported and I found no statements, whether in issues nor in ROS answers, about a working setup with ros_canopen and the effort controller.

However, your hint about the drive modes check from the EDS file was very helpful!
Thanks a lot.

ipa-mdl, 2021-09-02:
> The error message sounded to me like the effort interface is not implemented thus not loading

It is implemented, but not available, if the drive not support it.

> Also this table states, that torque control is not supported and I found no statements

The Schnunk robots do (did?) not support the (standard) torque control modes.
Elmo does, but I haven't tested them yet. That's why the support is "Unknown/?".