ros-industrial/ros_canopen #444: Unable to initialise controller with ICH Motion servo motors

stefano8302, 2021-09-15:
Hi everyone,
I am trying to integrate ICH Motion servo motors using ros_canopen with MoveIt on ROS Noetic running on Ubuntu 20.04, but I am unable to initialise the controllers to command a movement.
I will try to explain here below with more details what I am trying to do and how I have set everything up.
The motors I am employing aim at building a robotic arm that can be controlled by a PC running MoveIt on ROS.
The motors are 3-phase 24V brushless DC with a 17
[canconfig_stick3.txt](https://github.com/ros-industrial/ros_canopen/files/7171746/canconfig_stick3.txt)
 bit multi turn absolute encoder and rotor brake (then the motor will be coupled with a gearbox). The motor can be controlled by a driver which has also been provided by ICH Motion. The communication for the closed loop between the encoder and the driver is done via RS-485, while the connection between the driver and the controller/PC can be done via CANopen, RS-485, digital I/O, analog I/O and also USB if the manufacturer's diagnostics GUI is used.
Using the PC, the connection between the PC and the driver is done with a QinHeng Electronics HL-340 USB-Serial adapter (instructions to set it up found on https://github.com/alexmohr/usb-can).
The servo motors are not the ones advertised on ROS Wiki which are fully compatible and supported to be used with ROS using ros_canopen, but I believe that I have managed to adjust most of the settings to be able to control them, although I still have not managed to do it. The only controller (which I understand is not able to send commands) I have managed to initialise is the "joint_state_controller" and I can see the motor movements on RViz when I send commands form another source (the manufacture's GUI provided).
When I try to initialise any controller that is meant to send commands to the motor/robotic arm, I get the error that says it could not find the joint/joints in 'hardware_interface::EffortJointInterface' (this happens also if I try to initialise a PositionJointInterface or a VelocityJointInterface controller); so it fails to initialise/start the  controller.
The actual message is:
[ERROR] [1631696760.273566588, 554.016000000]: Could not find joint 'eje1_joint' in 'hardware_interface::EffortJointInterface'.
[ERROR] [1631696760.273651160, 554.016000000]: Failed to initialize the controller
[ERROR] [1631696760.274138060, 554.016000000]: Initializing controller 'joint_trajectory_controller' failed
[ERROR] [1631696760.274720584, 554.017000000]: Could not start controller with name 'joint_trajectory_controller' because no controller with this name exists.
 
I have followed these steps:
1. Socketcan is working
2. The URDF is OK
3. The .eds file has been provided by the manufacturer, but I had to revise and change it to make it work. Now I manage to initialise the driver with the ros_canopen  using canopen_motor_node.
4. The driver config for 1 motor/joint is as follows: 
[canconfig_stick3.txt](https://github.com/ros-industrial/ros_canopen/files/7171794/canconfig_stick3.txt)
5. list of controllers: 
[ros_controllers.txt](https://github.com/ros-industrial/ros_canopen/files/7171788/ros_controllers.txt)
6. The launchfile for canopen_motor_node:
[stick_can_ros_controllers.txt](https://github.com/ros-industrial/ros_canopen/files/7171823/stick_can_ros_controllers.txt)

When I use the launcher it seems to be correct and when I execute:
rosservice call /driver/init
I get a successful initialisation and I can see the CAN traffic in the candump.

If the controllers have not been loaded automatically, I use rqt controller manager (rosrun rqt_controller_manager rqt_controller_manager) to load and unload them for ease of use when debugging. When I try to load and start the joint_state_controller it is all good, but when I try to load the joint_trajectory_controller or eje1_joint_position_controller I get the error message above.

Please review my message and let me know if I have missed something that is useful to make you all understand the issue.

Thank you very much
regards,
Stefano

stefano8302, 2021-09-16:
I also attach the simple 1 axis URDF file I am using:
[stick3.txt](https://github.com/ros-industrial/ros_canopen/files/7175126/stick3.txt)

stefano8302, 2021-09-21:
Hi all,
I have done some more research and I wonder if there is a problem with the USB to CAN adapter (QinHeng Electronics HL-340 USB-Serial adapter).
I have reviewed the list of tested USB to CAN adapters on the page:
http://wiki.ros.org/socketcan_interface?distro=noetic
and thought that maybe the adapter I am using is not 100% compatible with socketcan_interface; possibly it is not able to work as a "master" but only as "slave"?
I am saying this also because I can only start the "joint_state_controller" which shows the joint position in real time on RViz, but any other "master" controller that sends orders to the motor cannot be loaded nor started. Thanks.

Regards,
Stefano

ipa-mdl, 2021-10-14:
It looks like your drive does not support the torque modes (4 or 10) or it does not report the support properly.
And in addition your configuration configures mode 1 (profiled position) or mode 7 (interpolated position) for your `effort_controllers/*` controllers.

----

This is clearly a wrong configuration and there does not seem to be any issue with `ros_canopen`, so I will close this now.
For more support, please try https://answers.ros.org/.

ipa-mdl, 2021-10-14:
> if my statement about not having just one of the two torque driver modes supported, generates the error and

As long as one torque mode is supported, `ros_canopen` will register the effort interface.
The URDF settings are only relevant for Gazebo..

> if the other error message I received is due to a wrong configuration or structure of the files in the package.

Wrong configuration.
https://github.com/ros-controls/ros_controllers/search?q=No+joint+given
Please note that the single-joint controller have a different config structure from the multi-joints one.