ros-industrial/ros_canopen #449: std::exception::what: Duration is out of dual 32-bit range

fzoric8, 2021-12-14:
I'm using `canopen_motor_node` as my driver for Schunk `LWA4p` robotic arm. 

Sometimes after sending motion plan to robot trajectory controller which uses ros control and mentioned driver to send motor commands over CAN to robot joints. 

Most of the time, everything executes correctly, however, sometimes I get following error: 
```
[ERROR] [1639496142.557002552] [/lwa4p/driver]: Dynamic exception type: std::runtime_error
std::exception::what: Duration is out of dual 32-bit range
```

And I'm not sure what can cause it. 
In meantime I will try to build `ros_canopen` from source, and my environment is: 

- Ubuntu 20.04 
- ROS noetic 
- SocketCAN driver 
- Docker version 20.10.3, build 48d30b5  

Any help would be appreciated. Thank you very much.

fzoric8, 2021-12-15:
I narrowed it down to following method: 

```
[ERROR] [1639561818.689072706] [/lwa4p/driver] [/home/developer/moveit_ws/src/ros_canopen/canopen_chain_node/src/ros_chain.cpp] [7fe035ffb700] [RosChain::run]: Dynamic exception type: std::runtime_error
std::exception::what: Duration is out of dual 32-bit range
```

fzoric8, 2022-01-12:
It would be great to find out how to debug `write` or `read` method. Thanks :)

OTL, 2022-01-17:
We have the same issue on Ubuntu20.04, noetic.

mathias-luedtke, 2022-01-17:
@ipa-hsd: Please try to confirm this issue with noetic?

mathias-luedtke, 2022-01-17:
@fzoric8, @OTL: This sounds similar to https://github.com/ros-planning/moveit/issues/1062

fzoric8, 2022-01-17:
> @fzoric8, @OTL: This sounds similar to https://github.com/ros-planning/moveit/issues/1062

Hi, I've seen that issue before. After that I've rebuilt my catkin workspace but error still persists. 

Above all, I'm using docker with new ros noetic as well as ros_canopen built from source. Still no luck, error persists. Therefore, I'm quite confident it's not due to the incompatible binaries.

mathias-luedtke, 2022-01-17:
> I'm using docker with new ros noetic as well as ros_canopen built from source

Not really a stable, well-defined system..

> My assumption in my case is that method read or write fail, and that causes this error although they're not related.

The read/write functions are the core of ros_canopen and glue all the different layers.
And any unhandled execption gets caught here to prevent the ROS node from crashing.

But ros::Duration (where you exception is from) is only used for dealing with ros_control(lers).

If it is not an ABI break it could be a division by zero (or something similar) as well. You could try to get a stacktrace.

To rule out any problem with ros_canopen itself: What gets logged from the following code?
https://github.com/ros-industrial/ros_canopen/blob/eaaf0a17b928db679215ddc9c3da1daaa6aa656c/canopen_motor_node/src/motor_chain.cpp#L71-L76

fzoric8, 2022-01-18:
Hi, out of curiosity, what would be defined as stable well-defined system? 

I'm currently not able to try it out, but after mid-february I'll give it a try and let you know.

miguelprada, 2022-02-03:
May also be related to ros-controls/ros_controllers#577.