ros-industrial/ros_canopen #452: How to use the canopen_motor_node?

nonlinear1, 2022-02-01:
I am a newer to this package. I have a servo driver with canopen protocol and a BLDC motor.
I think use this canopen_motor_node to control my motor. I am confused with the all kinds of setting including the ros_control. Could you give a detailed simple example or a detailed explanation? 
Thank you very much

mzahana, 2022-02-03:
Hi @nonlinear1 

I have the same question. I am also trying to interface the ZLAC3080L motor driver to ROS. The driver supports CANopen. However, there is no enough documentation for this `ros_canopen` package. So, I am not sure how to proceed.
Have you figured out a way to use this package?

Thanks.

nonlinear1, 2022-02-03:
Hi @mzahana , I am trying to solve it. I have found the following link: https://answers.ros.org/question/250174/how-to-control-maxon-motor-by-using-ros_canopen/ 
Maybe it will give you some help!
I think three configure files are needed: 1)urdf of your robot; 2)driver config files, that is this ros_canopen config files; 3) controller config files, that is the ros_control config files.

cschindlbeck, 2022-04-12:
I agree that documentation should be more verbose, maybe the README can be updated with minimal working examples or more specific instructions