ros-industrial/ros_canopen #455: Could not find the required component 'joint_limits_interface'

leeyunhome, 2022-03-11:
Hello

When I run catkin_make, I get the following error:
please tell me how to solve

-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found Boost: /usr/include (found version "1.65.1") found components: system 
-- Generating .msg files for action ros_tutorials_action/Fibonacci /home/manager/catkin_ws/src/ros_tutorials/ros_tutorials_action/action/Fibonacci.action
-- ros_tutorials_action: 7 messages, 0 services
-- +++ processing catkin package: 'canopen_motor_node'
-- ==> add_subdirectory(ros_canopen/canopen_motor_node)
-- Could NOT find joint_limits_interface (missing: joint_limits_interface_DIR)
-- Could not find the required component 'joint_limits_interface'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "joint_limits_interface" with any of the following names:

    joint_limits_interfaceConfig.cmake
    joint_limits_interface-config.cmake

  Add the installation prefix of "joint_limits_interface" to
  CMAKE_PREFIX_PATH or set "joint_limits_interface_DIR" to a directory
  containing one of the above files.  If "joint_limits_interface" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  ros_canopen/canopen_motor_node/CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/manager/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/manager/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Thank you.

cschindlbeck, 2022-03-11:
It appears that you are missing joint_limits_interface from ros-control. Try 

```
rosdep update
rosdep install --from-paths src --ignore-src -r -y
```
in your catkin_ws, the dependency is defined in canopen_motor_node package.xml

leeyunhome, 2022-05-04:
Hello, @cschindlbeck 
First, Thank you very much for your reply

I have a question.

When building ros_canopen on a machine called nvidia jetson nano,
I met the following error.
I'm stuck because I don't know hot to fix it.
Can you give me some advice?

I am attacking the error log as follows.
[CMakeError.log](https://github.com/ros-industrial/ros_canopen/files/8618824/CMakeError.log)
[CMakeOutput.log](https://github.com/ros-industrial/ros_canopen/files/8618825/CMakeOutput.log)
[ros_canopen_build_error_log_on_nvidia_jetson_nano.txt](https://github.com/ros-industrial/ros_canopen/files/8618815/ros_canopen_build_error_log_on_nvidia_jetson_nano.txt)

Thank you.

cschindlbeck, 2022-05-04:
it appears that you are missing the controller_manager, try

```sh
sudo apt-get install ros-melodic-ros-controller-manager
```

leeyunhome, 2022-05-04:
> it appears that you are missing the controller_manager, try
> 
> ```shell
> sudo apt-get install ros-melodic-ros-controller-manager
> ```

Hello @cschindlbeck 

I'll try it.

Thank you.

leeyunhome, 2022-05-04:
> it appears that you are missing the controller_manager, try
> 
> ```shell
> sudo apt-get install ros-melodic-ros-controller-manager
> ```

Hello @cschindlbeck 

manager@manager-desktop:~$ sudo apt-get install ros-melodic-ros-controller-manager
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-melodic-ros-controller-manager
manager@manager-desktop:~$ sudo apt-get install ros-melodic-ros
Display all 158 possibilities? (y or n)
manager@manager-desktop:~$ sudo apt-get install ros-melodic-ros-co
ros-melodic-ros-comm                        ros-melodic-ros-control-boilerplate-dbgsym  ros-melodic-ros-core
ros-melodic-ros-control                     ros-melodic-ros-controllers                 
ros-melodic-ros-control-boilerplate         ros-melodic-ros-controllers-cartesian       
manager@manager-desktop:~$ sudo apt-get install ros-melodic-ros-control

I can't find a package which name is ros-melodic-ros-controller-manager.
Can you tell me how to resolve this error?

Thank you.

cschindlbeck, 2022-05-04:
Can you try to install dependencies via rosdep?

```sh
rosdep update
rosdep install ros_canopen
```

leeyunhome, 2022-05-04:
> It appears that you are missing joint_limits_interface from ros-control. Try
> 
> ```
> rosdep update
> rosdep install --from-paths src --ignore-src -r -y
> ```
> 
> in your catkin_ws, the dependency is defined in canopen_motor_node package.xml

Hello, @cschindlbeck 

[  4%] Generating C++ code from can_msgs/Frame.msg
[  6%] Generating Python msg __init__.py for can_msgs
[  6%] Built target can_msgs_generate_messages_cpp
[  6%] Built target can_msgs_generate_messages_py
[  7%] Linking CXX shared library /home/manager/catkin_ws/devel/lib/libsocketcan_interface_string.so
[  7%] Built target socketcan_interface_string
[  9%] Linking CXX shared library /home/manager/catkin_ws/devel/lib/libsocketcan_interface_plugin.so
[  9%] Built target socketcan_interface_plugin
Scanning dependencies of target can_msgs_generate_messages_nodejs
[ 10%] Generating Javascript code from can_msgs/Frame.msg
[ 10%] Built target can_msgs_generate_messages_nodejs
Scanning dependencies of target socketcan_bcm
[ 12%] Building CXX object ros_canopen/socketcan_interface/CMakeFiles/socketcan_bcm.dir/src/canbcm.cpp.o
[ 13%] Linking CXX executable /home/manager/catkin_ws/devel/lib/socketcan_interface/socketcan_bcm
[ 13%] Built target socketcan_bcm
Scanning dependencies of target socketcan_dump
[ 15%] Building CXX object ros_canopen/socketcan_interface/CMakeFiles/socketcan_dump.dir/src/candump.cpp.o

I looked at the old records you told me and tried it.
It seems to be built.
I'll let you know when it's done.

Thank you.
Thank you.

leeyunhome, 2022-05-04:
> Can you try to install dependencies via rosdep?
> 
> ```shell
> rosdep update
> rosdep install ros_canopen
> ```

Hello, @cschindlbeck 

[ 96%] Linking CXX shared library /home/manager/catkin_ws/devel/lib/libcanopen_motor.so
[ 96%] Built target canopen_motor
Scanning dependencies of target canopen_motor_node
[ 98%] Building CXX object ros_canopen/canopen_motor_node/CMakeFiles/canopen_motor_node.dir/src/canopen_motor_chain_node.cpp.o
[100%] Linking CXX executable /home/manager/catkin_ws/devel/lib/canopen_motor_node/canopen_motor_node
[100%] Built target canopen_motor_node

I built it successfully thanks to you.

Thank you

cschindlbeck, 2022-05-04:
sure, can you close it then?

closed 2022-05-09 by leeyunhome